linuxboot/heads #195: add boot support for Tails

flammit, 2017-04-23:
Not sure if there's a desire to add more boot options before there's a more complex boot screen, but I've taken a shot at doing this for Tails.

There are a couple of boot options that are supported: 

1) a signed Tails ISO in the root of the drive (`/tails-i386-2.12.iso` and `/tails-i386-2.12.iso.sig`) 

2) the standard USB key made by Tails Installer.  

The signed ISO option requires that the `/tails-signing.key` be present as the signing key is quite large (450k).  The init script will expect the file be user-signed (like the `boot.hashes.asc`) or that it matches a hash in the script.

With the USB drive made from the Tails Installer, there's no good way to determine if the USB key has been tampered with, so there's a warning/confirmation on boot to proceed.

Will post a PR shortly.

osresearch, 2017-04-23:
This probably could be merged into #196 as a general way to select booting from external media. I like the idea of having the tails key included; is it possible to reduce the size by stripping the signatures to leave only the modulus?

flammit, 2017-04-23:
Yeah it's possible to reduce the key size.  You just need to import then `gpg edit-key` and `clean`.  I'll update to assume the Tails public key is in the ROM.

flammit, 2017-04-23:
As for merging into #196, I like it in concept but there are a couple issues I see.  

A generic ISO mounting routine would need to somehow deduce which magic kernel parameter is needed to boot properly (options from a quick Google seem to be at least `fromiso`, `isofrom`, `findiso`, `iso-scan/filename`).

Generic GRUB parsing could work, but you might need to add kernel parameters to boot properly and you might not know what ahead of time.  By debug/trial/error, it seemed like the `intel_iommu=on` is required for Tails to start properly.

flammit, 2017-04-26:
FYI - I had the same boot problems with Ubuntu live that are resolved with enabling iommu.  Might be related to this thread: https://lkml.org/lkml/2017/3/29/495

PR #200 (2017-07-17): adds a USB boot option with basic parsing to kexec
Supports booting from USB media using either the root device or
a signed ISO as the boot device.  Boot options are parsed with
quick/dirty shell scripts to infer kexec params.

Closes #195 and begins to address #196
commit efd662c6 (2017-07-17): adds a USB boot option with basic parsing to kexec

Supports booting from USB media using either the root device or
a signed ISO as the boot device.  Boot options are parsed with
quick/dirty shell scripts to infer kexec params.

Closes #195 and begins to address #196
closed 2017-07-17 by osresearch