linuxboot/heads #203: Encrypted home not mounted using passed keyfile

evq, 2017-06-07:
After following the installation instructions (minus read only root) with the master branch I am prompted on boot to put in the full disk recovery key to decrypt the home partition. It seems that after the encrypted root and swap are mounted by dracut execution immediately passes to the mounted root at `/sysroot`. While executing in the mounted root, the home partition tries to mount however it no longer has access to `/secret.key`.

I was able to fix this by creating a conf file in `/etc/dracut.conf.d/` with the following contents and then regenerating the initramfs:
```
add_device+= /dev/mapper/luks-<home_partition_id> "
```

Maybe I missed a setup step? Anyway, awesome project ^^

flammit, 2017-06-07:
The docs alluded to this issue (http://osresearch.net/Installing-Heads) but didn't provide a suggestion:

```TODO: how to force all of crypttab to be emitted?```

This looks much better than manually updating the initramfs to include a complete crypttab as I had been doing.

osresearch, 2017-06-07:
I mentioned the fix that I found on the qubes-devel mailing list, but haven't updated the docs. https://groups.google.com/d/msg/qubes-devel/ILsaXnKrL-M/zKbDEgVADAAJ
```

--- module-setup.sh.orig        2017-04-04 18:06:12.246999989 -0400 
+++ /usr/lib/dracut/modules.d/90crypt/module-setup.sh        2017-04-04 18:06:17.491999981 -0400 
@@ -75,7 +75,7 @@ 
  
             for _hdev in "${!host_fs_types[@]}"; do 
                 [[ ${host_fs_types[$_hdev]} == "crypto_LUKS" ]] || continue 
-                if [[ $_hdev -ef $_dev ]] || [[ /dev/block/$_hdev -ef $_dev ]]; then 
+                if true || [[ $_hdev -ef $_dev ]] || [[ /dev/block/$_hdev -ef $_dev ]]; then 
                     echo "$_mapper $_dev $_rest" 
                     break 
                 fi 
```

I like the idea of using the separate `/etc/dracut.conf.d/` entry, although it would require knowledge of the UUID.

flammit, 2017-06-08:
Ahh sorry I didn't see that qubes-devel message.

Until rd.luks.key works or this dracut patch, would you accept a PR to just have qubes-init automatically replace the initrd's crypttab at the same time it unseals the key?  The intent would be to match/mirror the seal-key logic and add entries for all volumes.

osresearch, 2017-07-17:
Fixed as part of #200 -- all devices in the qubes_dom0 group are added to the crypttab.

closed 2017-07-17 by osresearch