linuxboot/heads #204: Updates to coreboot-x220.config file are overwritten at build time

patrickod, 2017-06-08:
I'm currently trying to build an alterned coreboot-x220 target using an additional blob (`vgabios.bin`) and related configuration values. I've written the changes to `config/coreboot-x220.config` and included the new blob in the appropriate folder, however when I attempt a build using `make CONFIG=config/qubes-x220.conf` at some point during the build process the configuration file changes are reverted and are made ineffective. 

I'm not sure what's reverting the changes, but I noticed that the generated file `build/coreboot-4.5/x220/auto.conf` might be a candidate due to it containing the reverted values. However, it's not obvious to me how this auto.conf file would be copied back into the main heads repository.

osresearch, 2017-06-12:
As a test I edited `config/coreboot-x220.config` to change the version string and re-ran `make V=1 CONFIG=config/x220-qubes.conf` and verified that the resulting coreboot binary had the modified version.

This does re-run `make oldconfig` in the coreboot directory, which will modify the `.config` file.  If you have selected an invalid configuration (something where dependencies aren't available, etc) or if you have a typo, those lines will be removed.  Are you editing the file by hand or using the menuconfig?

```
make \
	-C build/coreboot-4.5 \
	DOTCONFIG=../../config/coreboot-x220.config \
	menuconfig
```

patrickod, 2017-06-18:
Hmm I was indeed editing the file by hand but will instead use the menuconfig option and see if it changes things. I'll also double check the paths referenced in my changes (specifically the addition of a VGA option ROM) to ensure that everything is in place. Thanks!

osresearch, 2018-03-16:
Is this still happening? Please re-open if so.

closed 2018-03-16 by osresearch