linuxboot/heads #208: black screen and stuck with ifd.bin from /blobs/x220

Kokokokoka, 2017-06-24:
Today I was trying to adapt latest coreboot with libgfxinit on x220 and it worked with heads setup!
Still the ifd.bin from blobs section almost gave me a headache as I was trying to understand what is wrong with my setup. And after almost the whole day of trying different builds I found out that it was purely wrong to use ifd.bin and not using/creating my own one. So maybe ifd.bin should be removed from blobs section? 
Other notes: libgfxinit works great. 
I bumped the kernel to the latest one (4.9.34) without any further changes.

osresearch, 2017-07-27:
It seems like the right thing to do is to update the instructions to extract the `ifd.bin` from the rom, like you suggest.  It's a little frustrating that `ifdtool` is part of coreboot, but the default file names for the output are not the same as the ones expected by the coreboot makefiles as input.

jgrip, 2017-08-04:
This should be resolved with #235 which implemented a script to extract the necessary parts from a flash dump and place them where they should be.

tlaurion, 2020-02-19:
Functional under https://github.com/osresearch/heads/commit/21faf524b9d791a75f1710ada502f21298b8fa77

closed 2020-02-19 by tlaurion