linuxboot/heads #215: Video glitches when kexec Ubuntu kernel

osresearch, 2017-07-16:
[image]

At the coreboot session during http://hackon.nl, @kliment found that his x230 would cause VGA display glitches when `kexec` into a stock 4.4.0-83 Ubuntu kernel.

It looks like @flammit solution to getting Tails to boot https://github.com/osresearch/heads/issues/195#issuecomment-296489243 might fix it.

merge, 2018-10-18:
I see the same glitches. This is __not__ related to Ubuntu. I run a stock 4.9 Debian kernel.

The thing is: The glitches last for as long as runtime-Linux hasn't set the fb-mode. After that, the display manager starts fine. However we don't have Linux console logs and this is a problem: If you use dm-crypt disk encryption, you are stuck and can't boot since Linux will stop and ask for the passphrase before setting modes!

`intel_iommu=on` is set, as in the master branch for the X230. I have https://github.com/osresearch/heads/pull/465 applied, but this should actually be unrelated, shouldn't it?

Any idea what else would make sense to test here?

merge, 2018-10-18:
I've found that I need `intel_iommu=on`. Having __only__ `intel_iommu=igfx_off` makes things worse and I'm stuck with the glitches forever.

merge, 2018-10-19:
> Are we talking about [this](https://github.com/osresearch/heads/pull/377/files#diff-2d4ed2860b55bb308eacc61e6c18a86fR640)? or [this](https://github.com/osresearch/heads/pull/452/files#diff-2d4ed2860b55bb308eacc61e6c18a86fR19) ? Those are required, with i915, to properly boot without screen flickers.
> 
sorry, but what's the differnce? isn't this the same thing in different pull requests? thanks!

tlaurion, 2018-10-20:
@merge : no difference. Just not present in the change subset being applied in #465 :)

tlaurion, 2018-10-22:
@merge : which version of Ubuntu were you able to install/ you attempt to use?
Not able to install [18.10 FTM](https://github.com/osresearch/heads/issues/470#issuecomment-431917443).

merge, 2018-10-23:
I don't use Ubuntu, only Debian 9. I didn't yet install it when running Heads, but my report is about booting an installed stock Debian kernel.

flammit, 2018-11-18:
@merge checkout this old commit: https://github.com/flammit/heads/commit/e5bad487e44d1eedf4a963273b9950abb04e01c2

As long as you set the coreboot/boot kernel to disable iommu (`intel_iommu=igfx_off`) on the graphics card and match it on the target/os kernel, you should be fine.

tlaurion, 2018-11-19:
@merge @flammit: updated #480 and #470.

merge, 2018-11-20:
thanks for all the review and testing! IMO this can be closed as soon as #465 or a similar change is merged that fixes this.

closed 2020-03-09 by tlaurion