linuxboot/heads #219: Heads recovery shell is not a nice video mode

osresearch, 2017-07-17:
How much time during the boot is necessary to initialize the video mode so that it uses the proper 8x16 font and native resolution?

osresearch, 2017-07-18:
Having the Heads kernel setup the video mode leads to glitching when Qubes starts up, and Xen no longer displays anything.

osresearch, 2017-07-18:
Maybe related to #215?

osresearch, 2017-07-18:
[image]

Looks great at the Heads recovery shell, but causes significant glitching.  `kexec` had some patches in the past for VESA framebuffers, but nothing for i915.  i wrote to the coreboot list about it: https://mail.coreboot.org/pipermail/coreboot/2017-July/084716.html

flammit, 2017-07-29:
Disabling IOMMU for the graphics in the Heads kernel (`intel_iommu=igfx_off`) stops a bunch of the early glitches before the new kernel's graphics starts up.  Setting that parameter in the target kernel clears up the rest, but there's still a bit of a hiccup (see here):

[image]

As for Xen, without any parameter changes it only has a minimal flash before starting up when Heads has `intel_iommu=igfx_off`.