linuxboot/heads #229: X230 generates heat under S3. Hard drives wakes up even when powered off?

tlaurion, 2017-07-20:
HD wakes up, even when powered off. Changed to spinning drive to sd drive resulting in lower battery drain.
ME is cleaned.

Am I alone having this issue?

osresearch, 2017-07-27:
I've noticed that my x230 seems to run the battery down in S3 as well.  I haven't investigated or compared to with the ME firmware, but there might be something weird in power management.

Do you find that your x230 won't shutdown correctly with the ME cleaned?

flammit, 2017-07-27:
FWIW I haven't had x230 shutdown problems with a cleaned ME.  I don't use sleep so I haven't run into that S3 battery issue.

tlaurion, 2017-07-28:
Shutdowns without any issue.

SamuelBowman, 2017-08-09:
I can confirm that I am also experiencing battery drain in S3. I can't confirm drive activity as I have an SSD in mine. I don't remember if I cleaned ME or not, but I do have a backup of the original Lenovo flash with ME so I can restore it and reflash heads to see if that helps.

If that doesn't fix it, I think it may be an issue with coreboot, but that's just a gut feeling. I might try flashing coreboot w/o heads and some different configs later to try and debug if ME isn't the issue. Hope this gets figured out soon as I miss my battery life.

merge, 2018-05-25:
FWIW, I installed coreboot on some x230 models since August 2017, and never have seen any S3 or shutdown issues.

tlaurion, 2018-07-27:
No more shutdown issues, but still heat generated with no HD complete shutdown while in S3.
@osresearch : Any troubleshooting paths or any known knowledgeable people that could give hints about this?

osresearch, 2018-07-27:
My x230 (with a cleaned ME) seems to not go fully to sleep -- if I leave it "sleeping" over night it has lost most of its power by the morning.  I'm not sure if this is a Heads issue or a coreboot issue, but there is definitely something going on with that machine.

SebastianMcMillan, 2019-03-20:
My X230t (runs X230 coreboot) running Tianocore + stock coreboot has battery drain issues in S3. I have a mSATA SSD, so I haven't noticed the drive being on, and there isn't any heat issues. Culprit is almost certainly coreboot.

tlaurion, 2021-07-29:
[s2idle instead of s3](https://github.com/systemd/systemd/issues/9538#issuecomment-510542919)