linuxboot/heads #234: Update docs to reflect new installation/upgrade procedures

flammit, 2017-07-29:
for reference, here's the old gist that describes the config files:  https://gist.github.com/flammit/fc77718651d59d720547dda6c544a04b

flammit, 2017-09-04:
`gpg` needs to be added to the package lists on http://osresearch.net/Building for validation in the qubes-vmm-xen build.

flammit, 2017-09-05:
Should reference the use of `x230-flash.rom` instead of `x230.coreboot.rom` in the installation procedure (http://osresearch.net/Installing-Heads)

osresearch, 2018-02-06:
Should we retain the `x230-flash.rom` build configuration?  On reflection I'm not sure if it is necessary -- most users will want to unlock their IFD and possibly run `me_cleaner`, so they will need to flash both SPI chips anyway.

tlaurion, 2019-01-27:
@flammit @osresearch @merge @kylerankin :
I think we should keep x230 and x230-flash.

Use x230-flash to flash a fbwhiptail version. And once #490 and #511 are in, the user will be able to own a new GPG card directly from the first boot, and flash a clean rom or upgrade Heads without losing previously added public key and trustdb.

This debate is linked with #307.

closed 2020-08-01 by tlaurion