linuxboot/heads #236: TPM out of resources in creating a counter

osresearch, 2017-08-14:
Following a full reinstall the TPM counter id is not known, so the signing script attempts to create a new one.  On the x230 only a single counter is supported (?), so this results in:

```
Got error TPM out of resources (0x15) from TPM_CreateCounter.
```

osresearch, 2017-08-14:
`tpm counter_release` should be able to delete the old counter, but the id, not the label, is necessary.  Is there a way to list counters?

flammit, 2017-08-14:
Yeah I would run into the same problem.  I couldn't find a way to list counters (by design?), so I would need to do a full TPM reset to clear the counters.  IIRC think it is more than one is supported (maybe 2 or 3 but I didn't check carefully).

osresearch, 2017-08-14:
Some sources say that the spec calls for a minimum of four, with only one modifiable per boot.  But nothing seems to say how to find the counter id, which seems like such an odd oversight.

So we should probably add a `tpm-reset` to the notes for a full re-install.

closed 2017-08-14 by osresearch