linuxboot/heads #237: Qubes 4.0 install failure

osresearch, 2017-08-14:
[image]

Not sure if this is Heads related -- I get an error about setting the default kernel and Qubes 4 doesn't start any VMs.  I'm retrying now without custom partitioning to see if that makes a difference.

osresearch, 2017-08-14:
Transcribing the error for future searchers;
```
['/usr/bin/qubes-prefs', 'default-kernel', '4.9.35-19'] failed
....
FileNotFoundError: (Errno 2) No such file or directory
```

Re-installing and selecting automatic partitioning didn't seem to change things..

osresearch, 2017-08-14:
Looks like this is a xen 4.6 versus 4.8 issue: https://github.com/QubesOS/qubes-issues/issues/3028#issuecomment-322330133

issue #159 is tracking the Ques R3.2 xen, not the newer R4.0.

flammit, 2017-08-14:
There's a PR to master for the updated R4.0 / xen 4.8 line (https://github.com/osresearch/heads/pull/210), but it probably should be merged to a separate branch in your main repo.  Maybe eventually becoming a config toggle?

osresearch, 2017-08-15:
I had missed that PR and am testing the 4.8 build now.  It seems like it should be configurable somehow since we'll need both 4.6 and 4.8 until Qubes R4 is stable.

osresearch, 2017-08-15:
It makes it further with 4.8.1, but `xenstored` still fails with `could not open /var/lib/xenstoredb`.  I'm going to try a full re-install again with the proper Xen hypervisor.

osresearch, 2017-08-15:
Ok, that worked.  The partial 4.6 install borked it, so a full reinstall with 4.8 fixed the problem.

closed 2017-08-15 by osresearch