linuxboot/heads #241: Problem with crosscompiler

Stifflersmum, 2017-09-04:
Using the make CONFIG=config/x230-qubes.config command results in the following error:
/home/user/heads/build/coreboot-4.5/util/cbfstool/lz4/lib/lz4frame.c:1092:33: error: this statement may fall through [-Werror=implicit-fallthrough=]

workaround:

 Add  /* fall-through */ in /home/user/heads/build/coreboot-4.5/util/cbfstool/lz4/lib/lz4frame.c    at line 1094.

dont know if it worked though, since my headbuild is not booting.

cheers

flammit, 2017-09-04:
I can try to repro, but can you let me know what tag/commit of Heads are you trying to build and what's your host OS & GCC version?

flammit, 2017-09-04:
This error looks like an upstream coreboot/cbfstool issue with GCC 7.

I confirm though that between Fedora 26 / GCC 7.1.1, Fedora 25 / GCC 6.3.1, Ubuntu 16.04 / GCC 5.4.0, I can reproducibly build the flammit/master branch, so once built it should work.

Stifflersmum, 2017-09-04:
the clone command was:

git clone https://github.com/osresearch/heads
gcc (Debian 7.2.0-1) 7.2.0
the os is Parrot os.(Debian)

cheers

flammit, 2017-09-04:
If you're building from the tip of https://github.com/osresearch/heads, you should get a rom with the following sha256 hash:

```
0c32532d9e247b2269fa18c46d4d4fc03015cef5542edc05547e5c0938b5c58d  x230.rom
```

I've tested this on my x230 and it boots properly.  Is this the hash you're getting? You will need your one-line workaround to build until this is integrated into the coreboot patches.

Stifflersmum, 2017-09-05:
Yes, thats the hashoutput i get:

$sha256sum x230.rom 
0c32532d9e247b2269fa18c46d4d4fc03015cef5542edc05547e5c0938b5c58d  x230.rom

Maybe im doing something wrong when i split the file?
here are the commands im using:

dd of=x230-4mb2.rom bs=1M if=x230.rom skip=8
dd of=x230-8mb2.rom bs=1M if=x230.rom skip=4

Stifflersmum, 2017-09-05:
I should mention, when im trying to boot, the boot-key lights up for a seccond, and thats it.Boots just fine from normal coreboot+Seabios+Me-cleaner though.

osresearch, 2017-09-05:
Are you flashing both chips?  If so you'll need into include an ME image and the flash descriptor section.  The `x230.rom` file doesn't include those (the `/bin/flashrom-x230` script only flashes the BIOS section and leaves the ME/FD alone)

Stifflersmum, 2017-09-05:
Yes im flashing both.If i flash only the 4mb one, (leaving the 8mb one like it was bevor, with ME_cleaner applyed) the power-key-light stays on, but the screen is still black.I could try flashing the unmodifed 8mb image, but i dont want ME on my computer.

flammit, 2017-09-05:
It might be easier to use the x230-flash.rom on the 4mb.

First externally flash your 8mb back to the working me-cleaned state.  Then build `make CONFIG=config/x230-flash.config`, and externally flash the resulting 4mb x230-flash.rom to the 4mb chip.

That should boot you to a prompt where you can then mount a USB drive containing the 12mb rom and use the /bin/flashrom-x230.sh from within the Heads to flash only the BIOS section as @osresearch indicated above.

Stifflersmum, 2017-09-06:
thanks, did that and worked fine as far as i can tell.

Stifflersmum, 2017-09-06:
Maybe someone cares to Help me: Im a bit stuck now, Heads is installed, when i boot up it shows !!!Bootscript missing? Entering recovery shell...
but when i plug in the usbstick with Qubes  and run "qubes-install" he doesnt recognize the command (not found)

cheers

flammit, 2017-09-06:
Yeah unfortunately some of the docs don't apply to tip anymore.  You should be using `config/x230-generic.config` not `config/x230-qubes.config` (which doesn't exist anymore) and the `CONFIG_BOOTSCRIPT` should be `/bin/generic-init` (which does exist).

1) After you flash the `x230.rom` and reboot (use the `reboot` command), you'll see a boot menu but it will drop to recovery failing to unseal the TOTP
2) Reboot again and then you can run `seal-totp`.  
3) Next reboot should have the TOTP code shown for verification.
4) Insert the Qubes USB, select 'u' for USB boot and choose the 'Install Qubes 3.2' to install
5) After installation, reboot then the 'd' default option or 'm' boot menu will give you the grub boot options for the installed Qubes.

Stifflersmum, 2017-09-07:
thanks, ill try that this weekend(need to resolder the clip first) and keep you updatet.If you want, i  can do an easy step by step tutorial for noobs like me  when i get HEADS to work.
btw, it doesent recognize (not found) the command "reboot", nore "halt", "poweroff" ore "shutdown -r now" 
which confuses me a litlle.

flammit, 2017-09-07:
Sorry for the confusion it's my mistake - the master branch on osresearch/heads doesn't have the reboot/poweroff commands merged yet.

Stifflersmum, 2017-09-08:
no harm done :).

Stifflersmum, 2017-09-22:
I forgot to Update: Everything is working fine know, Qubes installation went almost flawles.I didnt  replace the gpg key yet, since it was just a test-installation. Will Nitrokey instead of yubikey work also for this? Im asking cause Yubikey isnt OpenSource.

derlaft, 2017-11-07:
I have the same issue, can you please help me?

https://gist.github.com/8938a63a790911abf00248c4c1fa6744

flammit, 2017-11-24:
hey @derlaft - @Stifflersmum posted the workaround above (add the fall-through line in `lz4frame.c`), but if you still need help, I can add a patch for coreboot 4.6 (this is already fixed upstream in coreboot and shouldn't be a problem in 4.7).

commit 5f9567c3 (2017-12-11): Fix coreboot GCC7 build issue

This is fixed in coreboot master but backporting for Heads.

Closes #241
closed 2017-12-11 by osresearch