linuxboot/heads #252: NERF: test ME cleaner on r630

osresearch, 2017-09-20:
Can we free up more space and eliminate more vendor code?

citypw, 2017-11-27:
Have you test NERF on r730? I dumped a fw from r730 lately and the SPS version seems pretty much the same as r630 but I don't have r630 to test.

osresearch, 2017-11-27:
I have't tested on the r730.  If the ROM looks similar you can try building it.  It is best to dump the ACPI tables from a running system to include in the `blobs/r730/acpi` directory.

citypw, 2017-11-28:
OK, cu'z the flash chip is on the back of the mainboard so I'll have to reinstall the whole machine after reflash via external programmer each time. Any idea how to simplify the whole process? Otherwise, I'm going to convince my boss allow me to cut a small area off the metal on the chassis.

closed 2018-02-07 by osresearch