linuxboot/heads #273: No rule to make target: /path/to/heads/build//.canary

rotation3, 2017-10-13:
I am building Heads on a Musl-libc based system. (I do not get the following errors on any glibc-based system.)

When attempting to build, I get a rather cryptic error: 

make [1]: *** No rule to make target '/path/to/heads/build//.canary', needed by '/path/to/heads/install/include/linux/limits.h'. Stop.

I see that in a "canonical" build, (I did "make CONFIG=config/x230-flash.config" to start with), that when going about this normally, there is no install/ directory created straight after building the linux module (the last line before the error is "DONE linux").

How might we fix this? This seems to be a part of the build of the musl module. As far as I can tell, it should be doing other logic for building Musl before this. But it does not execute the logic above this step (probably accounting for the missing install/ directory).

I can in fact create the file in build/.canary. This results in the musl module building as usual. However, I think there must be something wrong with the way that the build system is checking for this file. I'm also not familiar with some of the caveats of the system, so I am a little worried I may be missing something crucial that would be a problem in the end (it does not seem to cause any problems to create it, though, as far as I can tell.)

I'd appreciate any possible insights into this matter.

duncanguthrie, 2017-10-13:
For some reason, we see that linux_dir is not being set correctly, so the musl module thinks this is empty. It can be hard-coded in modules/musl as a workaround, but I do not know why it isn't working.

osresearch, 2017-10-18:
I wonder if the missing `linux_dir` variable is due to a include ordering issue.  Most of the vars are set with `:=`, which requires that they exist before they can be used.  There is an ugly hack in `modules/musl` that does depend on the `modules/linux` file being included before it is included:

```
# Fake a target so that musl will force a header install by the
# Linux kernel sources.
musl.intermediate: $(INSTALL)/include/linux/limits.h
$(INSTALL)/include/linux/limits.h: $(build)/$(linux_dir)/.canary
        $(MAKE) \
                -C "$(build)/$(linux_dir)" \
                INSTALL_HDR_PATH="$(INSTALL)" \
                headers_install
```

osresearch, 2018-03-16:
Please re-open this if it is still an issue.

closed 2018-03-16 by osresearch