linuxboot/heads #286: NERF R630 - No IOMMU?

tlaurion, 2017-11-19:
I see no ticket opened for the absence of IOMMU noted as a "minor issue" [here](https://trmm.net/NERF#Installing_on_a_Dell_R630) : "The current status (2017-09-20) is that most things work, with the major exception of SMP and USB, and the minor exception of the IOMMU. ". 

Why is IOMMU broken?

Thanks

osresearch, 2017-11-20:
There is a kernel message/oops about DMAR being broken (reads as all zero?) in the ACPI table. I haven't debugged it yet -- we're using the ACPI tables from a system booted with UEFI, but I'm not sure if edk2 has any Vtd support. We could borrow `AcpiVtd.dxe` from the vendor ROM, although I'd prefer not to if we can figure out how to enable it ourselves.

osresearch, 2017-11-22:
```
[2017-10-23 09:08:16.318] [    0.000000] ------------[ cut here ]------------
[2017-10-23 09:08:16.322] [    0.000000] WARNING: CPU: 0 PID: 0 at drivers/iommu/dmar.c:844 warn_invalid_dmar+0x70/0x7d
[2017-10-23 09:08:16.331] [    0.000000] Your BIOS is broken; DMAR reported at address fbffc000 returns all ones!
[2017-10-23 09:08:16.340] [    0.000000] BIOS vendor: (null); Ver: (null); Product Version: (null)
[2017-10-23 09:08:16.348] [    0.000000] Modules linked in:
[2017-10-23 09:08:16.351] [    0.000000] CPU: 0 PID: 0 Comm: swapper Not tainted 4.9.38-heads #0
[2017-10-23 09:08:16.357] [    0.000000]  ffffffff81803d68 ffffffff8121c1aa ffffffff81803db8 0000000000000000
[2017-10-23 09:08:16.366] [    0.000000]  ffffffff81803da8 ffffffff81062d23 0000034c81803e28 0000000000000000
[2017-10-23 09:08:16.374] [    0.000000]  0000000000000000 0000000000000000 0000000000000000 0000000000000018
[2017-10-23 09:08:16.382] [    0.000000] Call Trace:
[2017-10-23 09:08:16.385] [    0.000000]  [<ffffffff8121c1aa>] dump_stack+0x4d/0x63
[2017-10-23 09:08:16.391] [    0.000000]  [<ffffffff81062d23>] __warn+0xb8/0xd3
[2017-10-23 09:08:16.396] [    0.000000]  [<ffffffff81062dfd>] warn_slowpath_fmt_taint+0x4e/0x6a
[2017-10-23 09:08:16.403] [    0.000000]  [<ffffffff818b0421>] ? __early_ioremap+0x13d/0x1b6
[2017-10-23 09:08:16.410] [    0.000000]  [<ffffffff812ed051>] warn_invalid_dmar+0x70/0x7d
[2017-10-23 09:08:16.416] [    0.000000]  [<ffffffff8149a80c>] dmar_validate_one_drhd+0x8e/0xa0
[2017-10-23 09:08:16.423] [    0.000000]  [<ffffffff812ecd14>] dmar_walk_remapping_entries+0xf1/0x14c
[2017-10-23 09:08:16.430] [    0.000000]  [<ffffffff818bb259>] detect_intel_iommu+0x61/0xf2
[2017-10-23 09:08:16.437] [    0.000000]  [<ffffffff8149a77e>] ? acpi_os_unmap_memory+0xb/0xb
[2017-10-23 09:08:16.443] [    0.000000]  [<ffffffff8189d6cd>] pci_iommu_alloc+0x50/0x6c
[2017-10-23 09:08:16.450] [    0.000000]  [<ffffffff818a6225>] mem_init+0x9/0x21
[2017-10-23 09:08:16.455] [    0.000000]  [<ffffffff81894ce6>] start_kernel+0x219/0x463
[2017-10-23 09:08:16.461] [    0.000000]  [<ffffffff8189428f>] x86_64_start_reservations+0x2a/0x2c
[2017-10-23 09:08:16.468] [    0.000000]  [<ffffffff818943b6>] x86_64_start_kernel+0x125/0x130
[2017-10-23 09:08:16.475] [    0.000000] ---[ end trace 8b5096dab479e9ab ]---
```

osresearch, 2018-02-07:
This is now LinuxBoot's problem (and appears to be working on at least s2600wf and Winterfell)

closed 2018-02-07 by osresearch