linuxboot/heads #298: gpg fails to find libpcsclite.so

duncanguthrie, 2018-01-13:
I am using the v0.2.1 release, and I attempted to use a Yubikey to sign the boot hashes.
Unfortunately, I have encountered a rather fatal error. I get the following message:

``gpg: apdu_open_reader: failed to open driver `libpcsclite.so': Error loading library libpcsclite.so: No such file or directory``
`gpg: card reader not available`
`gpg: OpenPGP card not available: general error`

This confuses me; gpg is built with the --enable-card-support option, so it has the internal CCID driver. However, I can't see how to force it to use the internal CCID driver, because that is the default.

tlaurion, 2018-08-03:
@duncanguthrie : do you still have the issue?

duncanguthrie, 2018-08-03:
I don't. I can't reproduce it anymore, either.

closed 2018-08-03 by duncanguthrie

tlaurion, 2018-08-04:
@osresearch : I'm having the error on a clean build.

duncanguthrie, 2018-08-04:
I am able to reproduce it if I have more than one smartcard reader installed - e.g. the Lenovo smartcard reader installed in the X230's ExpressCard slot and then insert the Yubikey. It might be because the Lenovo reader is not supported by old GnuPG, but I am not sure about this.

tlaurion, 2018-08-04:
@duncanguthrie 

> I am able to reproduce it if I have more than one smartcard reader installed - e.g. the Lenovo smartcard reader installed in the X230's ExpressCard slot and then insert the Yubikey. It might be because the Lenovo reader is not supported by old GnuPG, but I am not sure about this.
The library should or should not be there and should not result in this exact error. You sure it is complaining about not finding libpcsclite.so?

osresearch, 2018-08-05:
When the card is found, `libpsclite` is not used at all.  On my x230 I have to have the correct USB modules loaded (`xhci-hcd.ko`/`xhci-pci.ko` for the left ones, `ehci-hcd.ko`/`echi-pci.ko` for the right one), otherwise I get the exact same error.

Do the cards show up in `lsusb`? VID should be 1050, I think.

tlaurion, 2018-08-06:
USB drivers were not loaded. Thanks for the heads up, @osresearch  :)