linuxboot/heads #299: NERF: clean checkout of nerf branch fails to build

osresearch, 2018-01-16:
flashrom is wrong
musl has some http errors
edk2.patch barfs

c0d3z3r0, 2018-01-17:
[...]

build.py...
 : error 7000: Failed to execute command
	make tbuild [heads/build/edk2-UDK2017/Build/MdeModule/DEBUG_GCC5/X64/MdePkg/Library/BaseIoLibIntrinsic/BaseIoLibIntrinsic]

build.py...
 : error 7000: Failed to execute command
	make tbuild [heads/build/edk2-UDK2017/Build/MdeModule/DEBUG_GCC5/X64/MdePkg/Library/BaseLib/BaseLib]

build.py...
 : error F002: Failed to build module
	heads/build/edk2-UDK2017/MdePkg/Library/BaseIoLibIntrinsic/BaseIoLibIntrinsic.inf [X64, GCC5, DEBUG]

- Failed -
Build end time: 14:24:33, Jan.17 2018
Build total time: 00:00:14

make[2]: *** [Makefile:26: heads/build/edk2-UDK2017/Build/MdeModule/DEBUG_GCC5/X64/MdeModulePkg/Core/Dxe/DxeMain/DEBUG/DxeCore.efi] Error 1

osresearch, 2018-01-20:
Looks like gpg is also having problems on some machines -- it does not detect a cross compile and fails when it tries to run tests.

osresearch, 2018-02-06:
clean checkout on Ubuntu 16.04, building as root, with patch applied: gnupg still fails to detect cross compiling.

osresearch, 2018-02-06:
Now xen is failing. ```
make[2]: Entering directory '/home/build/heads/build/qubes-vmm-xen-4.6.6-35'
make[2]: *** No rule to make target 'xen.gz'.  Stop.
make[2]: Leaving directory '/home/build/heads/build/qubes-vmm-xen-4.6.6-35'
make[1]: *** [Makefile:285: xen.intermediate] Error 1
make[1]: Leaving directory '/home/build/heads'
Makefile:485: recipe for target 'all' failed
```

osresearch, 2018-02-06:
Looks like I shouldn't have removed `patches/xen-4.6.6-35.patch`.  Reverted:
https://github.com/osresearch/heads/commit/eb26a453619a1c484365fd80d79cff125d2bce12

osresearch, 2018-02-06:
LinuxBoot also required perl `Digest::SHA`.  Need to update instructions.

osresearch, 2018-02-07:
I THINK THIS IS FIXED.  Hurrah?

closed 2018-02-07 by osresearch