linuxboot/heads #302: Documentation on how to install on X230 needs to be updated

n13, 2018-01-22:
This documentation is fairly outdated:
http://osresearch.net/Installing-Heads

I followed the steps, flashed the Rom... and then... the instructions are pretty unclear as to what to do next. 

I followed some older instructions to get the TPM set up, that didn't work. (here: https://github.com/mfc/flashing-docs/blob/master/walkthrough%20for%20flashing%20heads%20on%20an%20x230.md)

I tried to run start-xen, that doesn't exist

There's instructions on how to get a patched Xen from USB, but not sure that is still needed (as per one closed issue)
There's instructions about using a Yubikey to set up a pgp key; but yubikeys are considered unsafe now so I'd like to use my own key. Also kind of confused as the key is needed in "initrd.cpio" file, but the yubikey generated result is then in initrd/.gnupg - not initrd.cpio. 

I did tpm-reset, which seemed to work.
I did seal-totp, which worked, but unseal failed with an error message; maybe because of the missing gpg key? 

I'd like to be able to run my existing Qubes install, or recover it somehow. Prior to flashing the ROM I had coreboot installed and was running Qubes 4 rc3. 

I'd be happy to document the process, but currently nothing is really working. It's late so I will try to recover my qubes install tomorrow.

Configuration
Hardware Thinkpad X230
ROM: coreboot, intel ME neutered with intel ME cleaner.
Software: Qubes 4 rc3
I cloned the github repo and did a build from that (latest as per yesterday). Build worked fine, I built x230.flash.rom and also the "make xen.intermediary" xen version.

osresearch, 2018-02-06:
The build is also going to change as a result of merging the nerf branch back into master (issue #305), as well as with the nearly ready Qubes 4.0 release.  We'll need to bump the Xen version to match, which I think #238 is tracking.

The unsafe Yubikeys were recalled and replaced, so recent ones should be ok: https://www.yubico.com/support/security-advisories/ysa-2017-01/

tlaurion, 2018-04-17:
@osresearch : I can take a lead on this.
But https://github.com/osresearch/heads/issues/375 needs to be fixed first!

osresearch, 2018-04-19:
#375 has been merged.  Thanks for taking the lead on docs!

merge, 2018-05-23:
Better docs are clearly needed. Doing tpm-reset and seal-totp works (normal boot and displaying a TOTP). Booting in an unsigned boot option (default Debian installation) doesn't work though. heads boots it, but the boot process fails as if the filesystem is corrupt or something. If this is not expected, I'll provide logs...  anyways. Please improve the docs :)

paulmenzel, 2018-05-23:
On 05/23/18 10:16, Martin Kepplinger wrote:
> Booting in an unsigned boot option (default Debian installation)
> doesn't work though. heads boots it, but the boot process fails as if
> the filesystem is corrupt or something. If this is not expected, I'll
> provide logs...
Thank you, but please in a separate issue. ;-)

merge, 2018-05-24:
@flammit Debian failing was actually a false alarm. lightdm failed (manually restarting was necessary), but then, gdm3 also recently fails for me during startup, and that's most likely a Debian [issue](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=890614).

osresearch, 2018-06-06:
https://github.com/osresearch/heads-wiki/pull/13 was merged, so I think this is resolved.

PR #424 (2019-02-08): x230: remove 4M and 8M split-images from the build
The bios regions of the 12M coreboot image is 7M: 4M and 3 of the 8M split
image. The rest of the 8M image _generated_ with fake data and not usable
on real systems! It's dangerous to create them and suggest flashing them
externally.

That's exactly why the x230-flash build target is there: To
have a self-contained 4M image and enable easy unlocking of the 8M image
using the _original_ data.

the heads-wiki project is updated accordingly.

Closes #307
Closes #302
commit 4ea17583 (2019-02-08): x230: remove 4M and 8M split-images from the build

The bios regions of the 12M coreboot image is 7M: 4M and 3 of the 8M split
image. The rest of the 8M image _generated_ with fake data and not usable
on real systems! It's dangerous to create them and suggest flashing them
externally.

That's exactly why the x230-flash build target is there: To
have a self-contained 4M image and enable easy unlocking of the 8M image
using the _original_ data.

the heads-wiki project is updated accordingly.

Closes #307
Closes #302
closed 2018-06-06 by osresearch