linuxboot/heads #307: x230 doesn't use IFD or ME sections

osresearch, 2018-02-07:
The 12 MB x230 ROM image isn't flashable since it doesn't include the IFD and ME sections.  This isn't normally a large problem for updates since `flashrom-x230` won't touch the ME region, but it does lead to problems for initial install.

merge, 2018-04-20:
This and #375 is really only one issue IMO :)

tlaurion, 2018-04-20:
I think it might a false problem.

For both x220 and x230 and other 2 spi chips boards, the IFD should be provided inside of Heads (./initrd/etc/x230-layout.txt), accompanied with a flash script (/initrd/bin/flashrom-x230.sh) for Heads to be able to flash internally, the IFD containing the ME region to not touch, though that region could be reduced a lot, no?

@osresearch : would flashtool  be able to deal with reading the IFD in flash dynamically and determine the real used ME region? I mean. On the x230, there is plenty of space when ME modules are trimmed and the free space is concatenated:
```
./me_cleaner.py ~/Documents/Firmwares/ME/x230/MeRelocated.rom -S  -O x230_me_cleaned.rom -r -t --extract-me extracted_me.rom
Full image detected
The ME/TXE region goes from 0x3000 to 0x500000
Found FPT header at 0x3010
Found 1 partition(s)
Found FTPR header: FTPR partition spans from 0xd00 to 0xcad00
ME/TXE firmware version 8.1.40.1416
Public key match: Intel ME, firmware versions 7.x.x.x, 8.x.x.x
The AltMeDisable bit is NOT SET
Reading partitions list...
 FTPR (0x00000d00 - 0x0000cad00, 0x000ca000 total bytes): NOT removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x2f)...
Reading FTPR modules list...
 UPDATE           (LZMA   , 0x04d207 - 0x04d3c5       ): removed
 ROMP             (Huffman, fragmented data, ~2 KiB   ): NOT removed, essential
 BUP              (Huffman, fragmented data, ~56 KiB  ): NOT removed, essential
 KERNEL           (Huffman, fragmented data, ~135 KiB ): removed
 POLICY           (Huffman, fragmented data, ~91 KiB  ): removed
 HOSTCOMM         (LZMA   , 0x04d3c5 - 0x05419f       ): removed
 RSA              (LZMA   , 0x05419f - 0x0593f5       ): removed
 CLS              (LZMA   , 0x0593f5 - 0x05eb8a       ): removed
 TDT              (LZMA   , 0x05eb8a - 0x065280       ): removed
 FTCS             (Huffman, fragmented data, ~18 KiB  ): removed
 ClsPriv          (LZMA   , 0x065280 - 0x065661       ): removed
 SESSMGR          (LZMA   , 0x065661 - 0x073f8b       ): removed
Relocating FTPR from 0xd00 - 0xcad00 to 0xd00 - 0xcad00...
 Adjusting FPT entry...
 Adjusting LUT start offset...
 Adjusting Huffman start offset...
 Adjusting chunks offsets...
 Moving data...
The ME minimum size should be 98304 bytes (0x18000 bytes)
The ME region can be reduced up to:
 00003000:0001afff me
Setting the AltMeDisable bit in PCHSTRP10 to disable Intel ME...
Extracting and truncating the ME image to "extracted_me.rom"...
Checking the FTPR RSA signature of the extracted ME image... VALID
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

tlaurion, 2018-04-20:
In this case:
```
00000000:00000FFF Descriptor     
00001000:00002FFF GBe                   
00003000:004FFFFF ME              
00500000:00BFFFFF BIOS
```

ME could be reduced to:`
00003000:0001AFFF ME`

Is there something i'm missing?

merge, 2018-04-20:
no, I think you're right. The question is mostly  whether we need the space and want to become dependent on me_cleaner being applied.

Am 20. April 2018 22:41:23 MESZ schrieb tlaurion <notifications@github.com>:
>In this case:
>```
>00000000:00000FFF Descriptor     
>00001000:00002FFF GBe                   
>00003000:004FFFFF ME              
>00500000:00BFFFFF BIOS
>```
>
>ME could be reduced to:`
>00003000:0001AFFF ME`
>
>Is there something i'm missing?

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

merge, 2018-05-23:
@osresearch yes the 12m isn't flashable which is dangerous when flashing externally (first-time install) and people get it wrong (and the [head wiki ](https://github.com/osresearch/heads-wiki/blob/master/Installing-Heads.md)might even be misleading a little).

FWIW, specifically (only) for the X230, there's now a tiny project named [Skulls](https://github.com/merge/skulls/tree/master/x230) that basically tries to hide as much as possible and simplify external flashing / unlocking the IFD (and applying me_cleaner). I think about adding a script to flash Heads from there too.

merge, 2018-11-22:
this comment actually belonges in here more: https://github.com/osresearch/heads/pull/424#issuecomment-441074700

tlaurion, 2018-12-29:
@osresearch @merge @flammit : 
The more I think about it, the more I start to believe [that the same approach as Purism](https://github.com/osresearch/heads/blob/master/blobs/librem_skl/get_blobs.sh#L29) should be taken to download and neutralize ME binary blob before inserting it into Heads's 12MB rom, with a valid IFD being generated for that ROM.

The x230 [flash.sh specifics could then overwrite the whole SPI instead of just the bios region](https://github.com/osresearch/heads/blob/master/initrd/bin/flash.sh#L14).

What do you think? Modifying the scripts.

tlaurion, 2019-01-02:
@merge @flammit @osresearch 

Missing some pieces of the [puzzle](https://github.com/tlaurion/heads/blob/x230_blobs/blobs/x230/get_blobs.sh).

merge, 2019-01-07:
why? what's the benefit when flashing internally on the system?

tlaurion, 2019-01-07:
@merge more space, the possibility to update ME internally following probable changes in the ROMP and BUP modules and to have a valid IFD instead of only being able to flash the bios region since the layout file is not provided anymore.  Flashrom 1.0 upgrade and related flash.sh modifications wiped it.

But basically, more usable space in SPI flash. Maybe a false need, again.

tlaurion, 2019-02-08:
@merge said in #424 which is now merge:
```
We need a plan here :) we don't even know what to do yet

    First decision to make: we should agree that generating the 4M and 8M images of the x230 board is wrong in any case. We'll never be able to safely generate a flashable thing of 12M. And there's no point in having the splits other than flashing externally, which is always wrong.
    Also, flashing externally is only done the first time. And that's what the x230-flash board is for, which makes the x230 4M/8M split even more misleading. But then, x230-flash is not sufficient. At least touching the other chip's IFD is necessary.

    So, second decision to make: Do we want to support "bootstrapping" explicitely inside of Heads?
        If so, let's add scripts that unlock the IFD on the one chip and flash x230-flash on the other, and guide users through all of it.
        If not, drop x230-flash completely and simply add "unlocked IFD" to the requirements and add a script or (even better?) live-USB-image that flashes x230 from outside of Heads, which makes "bootstrapping" quite neat.

thoughts?
```

I think the x230-flash should deal with it internally. If i get things right from skulls, it means that x230-flash misses ifd tool to unlock the IFD from within. @merge?

merge, 2019-02-11:
ifdtool's "unlock" has to be applied to the 8M chip's region. `x230-flash` (just like a skulls image) uses 4M (the 4M chip) and thus makes it possible to hardware-flash "read, unlock, write" to the 8M chip, which makes the process as easy as possible (as of now). That documentation is what's important.

x230-flash shouldn't include ifdtool in initrd. For [future write-protection solutions](https://coreboot.coreboot.narkive.com/ovqZj0CI/spi-controller-and-lock-bits) we won't use IFD. We just need it to be unlocked once.

tlaurion, 2019-05-22:
Interesting points posted in #564 

@merge 
>  I haven't really looked into the current state of the x230-flash "board". IMO it shouldn't need gpg at all. It's for bootstrapping only, right?

gpg is not there anymore, since GPG2 is too big and x230-flash is used only to bootsrap.
@shamen123 : the 12mb, internally flashed version is used for everything else then initial flashing of Heads.

> Once the 4M "board" itself is in a nice shape (flash-gui, ... can be tested independently of the bootstrapping problem), we can think about how to help users with IFD unlocking. I can imagine a similar solution to https://github.com/merge/skulls/blob/master/x230/external_install_bottom.sh and good documentation. That script can directly be used actually... If it wasn't for "x230-flash" (a flash-only solution independent of the hard disk), we could even say "install Skulls first, and then come back here and flash Heads for your running Linux distro", but it's of course nicer to have a flash-only solution.

Unfortunately, following #517 and 4mb limit, having a flash-gui displayed in FBWhiptail or Whiptail only, won't fit in 4Mb.

I will look again on the Skull's external_install_bottom.sh script, since I can't still wrap my head about how we could deal with this one shot from the external, flashing system. Never actually played with Skull. Will try to find some time.

@shamen123 
>  i did make write up notes at the time and they are lurking around on one of my machines somewhere, which may be of use to others. but that was based on 0.2.1 generic init, so its likely already out of date.

The process hasn't change a lot since that time. Chip that script in?

merge, 2019-05-23:
> Interesting points posted in #564
>
> @merge
>
>
> gpg is not there anymore, since GPG2 is too big and x230-flash is used only to bootsrap.
> @shamen123 : the 12mb, internally flashed version is used for [...]

not yet at least :) but alright. There's no need to have a GUI for that one job.

> 
> I will look again on the Skull's external_install_bottom.sh script, since I can't still wrap my head about how we could deal with this one shot from the external, flashing system. Never actually played with Skull. Will try to find some time.

It's nothing special. Just a different 4M coreboot build, and thus it has all the bootstrapping-benefits that we need equally need to take advantage of for "x230-flash" here. Basically, you can even directly replace a pre-built "skulls" image with an "x230-flash" image and directly use skulls' "external" scripts - one per flash chip.

> 
> @shamen123
> 
> > i did make write up notes at the time and they are lurking around on one of my machines somewhere, which may be of use to others. but that was based on 0.2.1 generic init, so its likely already out of date.
> 
> The process hasn't change a lot since that time. Chip that script in?

tlaurion, 2019-10-28:
@merge : There is not enough space anymore in the BIOS region as is.

Playing around and applying [this](https://github.com/corna/me_cleaner/wiki/Internal-flashing-with-coreboot#neutralize-and-shrink-intel-me), including ifd and me in rom (while modifying flash.sh for it to reflash whole 12mb, not just BIOS region, there is now around 5mb flash that just appeared magically.

Looking at how Purism distributes its binaries through whole [rom download for their board](https://source.puri.sm/coreboot/releases/raw/master/librem_13v4/coreboot-heads-l13v4.rom.gz) there doesn't seem to be any problem redistributing ifd and me, which are basically included in the downloadable image, and extracted by their scripts.

@kylerankin ? @zaolin :  Would there be a legal problem distributing me.bin (98kb shrinked bin) and resulting ifd.bin to be included in x230 builds directly from the blobs directory on the repo?

What I propose:
- Create x230 blobs directory.
- Put cleaned_me.bin and reduced ifd.bin there
- Modify coreboot config so it builds with those dependencies included.
- Remove x230-flash (unneeded. There will never have enough place in 4Mb to do something useful there anyway.)
- Modify x230 board config to generate top.rom(4mb) and bottom.rom(8mb) ROMs.

This way, reading roms is useful only to have backups of them, but there is no real need to even run me_cleaner anymore. Just download GitlabCI artifact, and flash.
x230 me_cleaner command generating blobs would ressemble this:
`python me_cleaner.py -S -r -t -d -O out.bin -D ifd_shrinked.bin -M me_shrinked.bin original_down_dump.bin`

What do you think? Playing around to include thin-provisioning-tools right now to measure OEM distributed image directly from distributed /boot with dm-verity at reownership. More space is needed and my [quest to reduce firmware size](#590 ) has not gone very far. 
@marmarek: That would be useful too if fwupd subproject goes farther to provide firmware upgrades from QubesOS over sys-whonix from dom0.

osresearch, 2019-11-12:
The `ifd.bin` should be included; I don't think there are any licensing issues.  `me.bin` I'm not sure about.

Intel has a very reasonable [redistribution clause for their microcode updates](https://github.com/intel/Intel-Linux-Processor-Microcode-Data-Files/blob/master/license) that came about from [the Debian pushback](https://www.theregister.co.uk/2018/08/23/intel_microcode_license/) as well as [for the FSP](https://github.com/IntelFsp/FSP/blob/master/FSP_License.pdf).

However, the ME team has not had any similar sort of large scale push them to change their license. Lots of us have asked, but I've never received a response.

Fetching it from a separate blobs repo would at least avoid tainting the main Heads tree, which seems to be the coreboot approach.  This also avoids bloating up the tree with how ever many devices end up being supported.

merge, 2019-11-25:
> @merge : There is not enough space anymore in the BIOS region (coreboot cbfs defined space) as it is to do anything else.
> 
> Playing around and applying [this](https://github.com/corna/me_cleaner/wiki/Internal-flashing-with-coreboot#neutralize-and-shrink-intel-me), including ifd and me in rom (while modifying flash.sh for it to reflash whole 12mb, not just BIOS region), there is now around 5mb flash that just appeared magically.
> 

I guess we should just do that. Might be the easiest way to keep the x230 supported. I might think that through sometime... (we need more space for vboot later anyways).

(not now but for the future, I'd be interested: would we then need a new vboot fmd file in coreboot too? how would that look like?)

merge, 2019-11-28:
I tested this too and it works nicely: https://github.com/merge/heads/commit/4cbab95691d169aa4d7e52f27717be5a5a7fddcb 5M of free space. we need to be careful how to make the transition:

We change the IFD, so flashing from inside of Heads won't (conveniently) work! I guess before we do a Heads-release, we can and should do this and force flashing from outside of Heads (ifd, me and bios regions (we can write a script) from userspace or 4M/8M from external hardware).

do you see any way we could avoid that "break" for users? IMO the state of the project would allow us to do it and say "please flash from outside of Heads, once now", when doing a release?

tlaurion, 2019-11-28:
#590 can be closed when this is resolved.

tlaurion, 2020-01-26:
@osresearch @merge : In the goal of deploying descriptor.bin and me.bin, I was wondering what would be the best method to do this. Patch coreboot-blobs with additional files? Download github archive from another repo for a specific commit ID and extract into `mainboard/lenovo/x230/`?
```
[user@heads-x230-libremkey heads]$ find build/coreboot-4.8.1/3rdparty/ -name me.bin -o -name descriptor.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/google/stout/descriptor.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/google/stout/me.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/google/parrot/descriptor.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/google/parrot/me.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/google/link/descriptor.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/google/link/me.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/google/butterfly/descriptor.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/google/butterfly/me.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/samsung/stumpy/descriptor.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/samsung/stumpy/me.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/samsung/lumpy/descriptor.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/samsung/lumpy/me.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/intel/emeraldlake2/descriptor.bin
build/coreboot-4.8.1/3rdparty/blobs/mainboard/intel/emeraldlake2/me.bin
```

merge, 2020-01-27:
> @osresearch @merge @kylerankin : In the goal of deploying descriptor.bin and me.bin (and other blobs), I was wondering what would be the best method to do this globally for all boards.
> 
>     * Put a download script under blobs/x230 that fetches the binaries and store them locally in that directory (current way).
> 
>     * Patch coreboot-blobs with additional files (see below)?
> 
>     * Download github archive from another repo for a specific commit ID and extract into `mainboard/lenovo/x230/`?
> ```

go with the first, current way because that's straight forward and only simple additions. Moving the extract/download scripts out of the way in favour of putting stuff in coreboot's blobs repo is a different task that would make sense for all boards, but doesn't have to block this problem: Create a patch to the coreboot blobs repo, put it up for review and in the meantime, include it as a patch in Heads that includes all the binaries.

tlaurion, 2020-01-29:
Best way for newer coreboot versions seems to be to have them included directly into coreboot-blobs, IMOHO.

tlaurion, 2020-02-09:
@BrianRichardsonIntel: any input on ME redistribution, or more precisely https://github.com/osresearch/heads/issues/307#issuecomment-552961256 ?

tlaurion, 2020-02-19:
@merge 
> go with the first, current way because that's straight forward and only simple additions. Moving the extract/download scripts out of the way in favour of putting stuff in coreboot's blobs repo is a [...]

@MrChromebox : The problem with this approach is that no Continuous Integration platform will produce those boards reproducible artifacts for users. Another approach is needed.

MrChromebox, 2020-02-19:
> I may be rogue in my ways a bit, but personally, I would simply put the neutered blobs online and wait for Intel to ask to remove them, which would start talking about the issue for real

100% agree

tlaurion, 2020-02-19:
@MrChromebox : so let's do it?

merge, 2020-02-26:
> > I may be rogue in my ways a bit, but personally, I would simply put the neutered blobs online and wait for Intel to ask to remove them, which would start talking about the issue for real
> 
> 100% agree

I also agree here.

tlaurion, 2020-03-05:
@osresearch : thoughts?

tlaurion, 2020-03-09:
@osresearch @merge @kylerankin @MrChromebox :
The same issue is repeating itself over and over again for all to be supported boards if we want to have other tools included in Heads, latest being #690.

We have to agree on a plan. Go!

SebastianMcMillan, 2020-03-09:
> > > I may be rogue in my ways a bit, but personally, I would simply put the neutered blobs online and wait for Intel to ask to remove them, which would start talking about the issue for real
> > 
> > 
> > 100% agree
> 
> I also agree here.
Count me in as well.

tlaurion, 2020-03-09:
Coreboot refuses to add blobs in coreboot-blobs without license, it [seems](https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/thread/WRPJSARJASXP5GRGBCIKNM7LHE5PEMFS/)

tlaurion, 2020-03-10:
The more I think about it, the more I think we have to seperate boards from now on.
This, for example, could be x220-libremkey t420-libremkey, and should point to proper neutering instructions placing well carved ifd and me blobs to build coreboot for a specific Lenovo ROM update. 

Maybe the way to go is to actually borrow/rewrite code from @1vyrain to extract ME from original Lenovo BIOS upgrades directly, and patch Makefile/coreboot-blobs script in Heads so that when CONFIG_NEUTERED_ME is explicitely stated in board config, board blob scripts are called to download those original roms, apply me_leaner the way we need it and drop ifd.bin and me.bin and other needed blobs locally, to be able to build Heads from CI instances and [have reproducible builds](https://github.com/osresearch/heads/issues/571)....

eganonoa, 2020-03-11:
@tlaurion 
> The more I think about it, the more I think we have to seperate boards from now on.
> 
> #693 , for example, could be x220-libremkey t420-libremkey, and should point to proper neutering instructions placing well carved ifd and me blobs to build coreboot for a specific Lenovo ROM update.

As someone who struggled initially to figure out how to get the Nitrokey to work with Heads (in part confused by the different naming conventions: HOTP/LibremKey/Nitrokey), I think having separate boards for Heads with a key and Heads without a key (for all systems) would make a ton of sense. That would avoid people having to mess around with the various .config files and would just allow them to _make BOARD=x220_ or _make BOARD=x220_HOTP_ (or something like that). It would also allow for a simple addition to README.md and would help you figure out who can and who cannot do the testing of how well the keys work on different systems. Potentially it would also be useful for Purism and Nitrokey to point to the right way to do this in their own manuals for the Librem Key and the various compatible Nitrokeys?

tlaurion, 2020-03-11:
Searching ways to actually get the rom images directly from [Lenovo](http://forums.mydigitallife.info/threads/13194-Tool-to-Insert-Replace-SLIC-in-Phoenix-Insyde-Dell-EFI-BIOSes?p=196612&viewfull=1#post196612) and go from there.

[Adventure](https://www.coreboot.org/Bios_extract_Lenovo) in the goal of putting those blobs in total legaility in a CI:

- [ ] innoextract can extract FL2 file.
- [ ] [phcomp.exe ](http://www.endeer.cz/bios.tools/phcomp.exe) cannot run in dosbox to extract flh files (Actual BIOS)
- [ ] [e_bcpvpw.exe](https://www.mediafire.com/file/s1gw22ugnb125v9/e_bcpvpw.7z/file) cannot run in dosbox to extract flh files (Actual BIOS)
- [ ] me_cleaner should be able to run on extracted image to extract, neuter and expend IFD and dump directly under blobs directly.

tlaurion, 2020-03-12:
Most interesting project that *could* do the job so far seems to be [uefi-firmware-parser](https://github.com/theopolis/uefi-firmware-parser), which supports FL2 with bruteforce (-b) statement, but non-working. me_cleaner doesn't pick up on any complete 8MB file on which it can work on:
`wget https://download.lenovo.com/pccbbs/mobiles/g2uj31us.exe && innoextract ~/g2uj31us.exe && echo -e "\n\n EXTRACTING FL1\n\n\n" && /usr/bin/python2.7 /usr/local/bin/uefi-firmware-parser ./app/G2ETB5WW/\$01D3000.FL1 -e && echo -e "\n\n\n EXTRACTING FL2 \n\n\n" && /usr/bin/python2.7 /usr/local/bin/uefi-firmware-parser ./app/G2ETB5WW/\$01D3000.FL2 -e --superbrute && echo -e "\n\n\n Testing me_cleaner against all extravted files.... output given only on sucessful files... \n\n\n" && find ./pfheader* ./capsule-Capsule/ ./volume-0/ -type f | while read filename; do python ~/me_cleaner/me_cleaner.py -S -r -t -d -O out.bin -D ifd_shrinked.bin -M me_shrinked.bin $filename &>/dev/null; if [ $? -eq 0 ]; then echo $filename; fi; done;`

Nor match any file that would me the ME envelope to work on on the more generic me_cleaner command:
`wget https://download.lenovo.com/pccbbs/mobiles/g2uj31us.exe && innoextract ~/g2uj31us.exe && echo -e "\n\n EXTRACTING FL1\n\n\n" && /usr/bin/python2.7 /usr/local/bin/uefi-firmware-parser ./app/G2ETB5WW/\$01D3000.FL1 -e && echo -e "\n\n\n EXTRACTING FL2 \n\n\n" && /usr/bin/python2.7 /usr/local/bin/uefi-firmware-parser ./app/G2ETB5WW/\$01D3000.FL2 -e --superbrute && echo -e "\n\n\n Testing me_cleaner against all extravted files.... output given only on sucessful files... \n\n\n" && find ./pfheader* ./capsule-Capsule/ ./volume-0/ -type f | while read filename; do python ~/me_cleaner/me_cleaner.py $filename &>/dev/null; if [ $? -eq 0 ]; then echo $filename; fi; done;`

Goal here would be to extract 8Mb SPI to be flashed file, apply me_cleaner (`python me_cleaner.py -S -r -t -d -O out.bin -D ifd_shrinked.bin -M me_shrinked.bin original_down_dump.bin`) to extract+neuter+deactivate ME and recreate expended IFD to be dropped in blobs/x230 to be useable by remote public CIs to produce reproducible builds from ROM upgrades downloaded directly from manufacturers, to mitigate legal issues and manual scripts like presently enforced from the blobs/boards directories.

@Mrchromebox? @osresearch? @theopolis? @merge @n4ru?

MrChromebox, 2020-03-12:
I feel like this needlessly complicates the problem due to it's rigidity, fragility, and needless repetition of static tasks. If we wanted to have this as a manual script that feeds a blobs submodule, then I'd say fine, but do we really want to be downloading from Lenovo every simple time we build a Thinkpad board? And twice if building variants with and without a HOTP key? And how does this extend to other boards?

Purism originally did this with their coreboot build script, and ultimately there was lots of breakage due to sites outside of Purism't control going down, changing URLs, etc. Then we shifted to simply downloading/extracting from one of our own images.  Currently I'm moving everything to a blobs submodule, because it's easier/cleaner for both us and our users.

tlaurion, 2020-03-13:
@mrchromebox: the issue Here for x220 x230 t420 t430 and coming t530 all depends on Lenovo distribution licensing agreement to permit redistribution of ME blobs, which they haven't prohibited nor permitted. 

Thinkpad-ec project like many others, download iso or IMG and patch it directly from constant Lenovo URLs that haven't changed for years.

Coreboot-blobs dont want to include new blobs without accompanying license from manufacturer. So from my understanding, Purism is in position of distributing the blobs in coreboot-blobs if they'd like, but there is no clear stance for heads. 

I don't see any legal problem with this solution, while others are less clear.

For CIs, those binaries, like archive file or git archive downloaded by heads buildsystem, files would be downloaded only once per initial build, and each CI would reuse downloaded archive until a commit id changes in modules file, where newer archive and patches would be applied and the build system would reuse chache from there.

So only the first board build would download from Lenovo.

Of course if you have any information which would confirm the non-issue on deploying ME neutered blobs and expended IFD directly under Heads, that would change the whole situation. Else it currently requires the blobs to from an external backup of a ROM. 

If an external process permits to  reproduce the same verification hash from an external source, we could also distribute the blobs IMHO, without having to have heads download from Lenovo. But the script provided in the blobs directory should permit the user to arrive exactly at the same hash measurements, without having to trust any third party other then the original manufacturer's released firmware.

Any other idea?

tlaurion, 2020-03-13:
> Purism originally did this with their coreboot build script, and ultimately there was lots of breakage due to sites outside of Purism't control going down, changing URLs, etc. 

@MrChromebox : Originally, Purism was downloading from [mediafire](https://github.com/osresearch/heads/commit/f20d45a42638b11d7ca046b0e379150e3e932cde#diff-a5a01af914be455088da3edab3c13259) for blobs that were extracted from flashed roms, backuped, and then extracted.

> Then we shifted to simply downloading/extracting from one of our own images. 
Exactly. That would be the equivalent of downloading original BIOS files from Lenovo/Manufacturers, from original non-changing URLs, and applying recipes to be able to extract consistently blobs from those original images, but automatically and applying neutering recipes.
>Currently I'm moving everything to a blobs submodule, because it's easier/cleaner for both us and our users.

tlaurion, 2020-03-13:
Trying the chipsec path.

n4ru, 2020-03-14:
I’ll take a look at this. The offsets are identical generally across the entire generation, so once you find them for one chipset they’ll work to extract from the entire line.

tlaurion, 2020-03-14:
@n4ru: you got it right that is exactly the goal, having Lenovo images downloaded and 8mb firmware rom (including IFD, ME) extracted so that me_cleaner can be ran against it in the case of x230, and extending to other boards from there with fixed tools.

tlaurion, 2020-03-14:
Functional test should be :

python me_cleaner.py -S -r -t -d -O unneded_me.bin -D ifd_expended.bin -M me_shrinked.bin externally_extracted_rom.bin 

Where  ifd_expended.bin and me_shrinked.bin are the desired files.

n4ru, 2020-03-15:
Here you go.

[image]

[Download](https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t430/downloads/DS032435) and extract to the Drivers folder. In there, `ME8_5M_Production.bin` is the file. However, it is NOT a UEFI module, and I'm not sure if this is enough, as it is the binary itself, whereas normally UEFI modules have some "wrapping" when written to the BIOS chip itself.

tlaurion, 2020-03-15:
@n4ru : 

> ME/TXE image detected -d, -D, -M, -S and -s require a full dump

This is why finding a way to extract [ME region from full ROM](https://download.lenovo.com/pccbbs/mobiles/g2uj31us.exe) would be better, extracting minimal ME size and expended IFD in one shot, reproducibly.

tlaurion, 2020-03-15:
@n4ru 

Well, good news:
`wget https://download.lenovo.com/pccbbs/mobiles/g1rg24ww.exe && innoextract g1rg24ww.exe && python ~/me_cleaner/me_cleaner.py -r -t -O heads/blobs/x230/extracted_me.bin app/ME8_5M_Production.bin`

```
ME/TXE image detected
Found FPT header at 0x10
Found 23 partition(s)
Found FTPR header: FTPR partition spans from 0x180000 to 0x24a000
ME/TXE firmware version 8.1.72.3002 (generation 2)
Public key match: Intel ME, firmware versions 7.x.x.x, 8.x.x.x
Reading partitions list...
 ???? (0x000003c0 - 0x000000400, 0x00000040 total bytes): removed
 FOVD (0x00000400 - 0x000001000, 0x00000c00 total bytes): removed
 MDES (0x00001000 - 0x000002000, 0x00001000 total bytes): removed
 FCRS (0x00002000 - 0x000003000, 0x00001000 total bytes): removed
 EFFS (0x00003000 - 0x0000df000, 0x000dc000 total bytes): removed
 BIAL (NVRAM partition, no data, 0x0000add0 total bytes): nothing to remove
 BIEL (NVRAM partition, no data, 0x00003000 total bytes): nothing to remove
 BIIS (NVRAM partition, no data, 0x00036000 total bytes): nothing to remove
 NVCL (NVRAM partition, no data, 0x00010511 total bytes): nothing to remove
 NVCM (NVRAM partition, no data, 0x0000493f total bytes): nothing to remove
 NVCP (NVRAM partition, no data, 0x0000a553 total bytes): nothing to remove
 NVJC (NVRAM partition, no data, 0x00004000 total bytes): nothing to remove
 NVKR (NVRAM partition, no data, 0x0001257d total bytes): nothing to remove
 NVOS (NVRAM partition, no data, 0x00034af7 total bytes): nothing to remove
 NVSH (NVRAM partition, no data, 0x00007609 total bytes): nothing to remove
 NVTD (NVRAM partition, no data, 0x00001eac total bytes): nothing to remove
 PLDM (NVRAM partition, no data, 0x0000a000 total bytes): nothing to remove
 GLUT (0x000df000 - 0x0000e3000, 0x00004000 total bytes): removed
 LOCL (0x000e3000 - 0x0000e7000, 0x00004000 total bytes): removed
 WCOD (0x000e7000 - 0x000140000, 0x00059000 total bytes): removed
 MDMV (0x00140000 - 0x000180000, 0x00040000 total bytes): removed
 FTPR (0x00180000 - 0x00024a000, 0x000ca000 total bytes): NOT removed
 NFTP (0x0024a000 - 0x0004a4000, 0x0025a000 total bytes): removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xed)...
Reading FTPR modules list...
 UPDATE           (LZMA   , 0x1cc508 - 0x1cc6c6       ): removed
 ROMP             (Huffman, fragmented data, ~2 KiB   ): NOT removed, essential
 BUP              (Huffman, fragmented data, ~56 KiB  ): NOT removed, essential
 KERNEL           (Huffman, fragmented data, ~135 KiB ): removed
 POLICY           (Huffman, fragmented data, ~91 KiB  ): removed
 HOSTCOMM         (LZMA   , 0x1cc6c6 - 0x1d343f       ): removed
 RSA              (LZMA   , 0x1d343f - 0x1d872a       ): removed
 CLS              (LZMA   , 0x1d872a - 0x1ddec0       ): removed
 TDT              (LZMA   , 0x1ddec0 - 0x1e45be       ): removed
 FTCS             (Huffman, fragmented data, ~18 KiB  ): removed
 ClsPriv          (LZMA   , 0x1e45be - 0x1e499f       ): removed
 SESSMGR          (LZMA   , 0x1e499f - 0x1f32cb       ): removed
Relocating FTPR from 0x180000 - 0x24a000 to 0xd00 - 0xcad00...
 Adjusting FPT entry...
 Adjusting LUT start offset...
 Adjusting Huffman start offset...
 Adjusting chunks offsets...
 Moving data...
The ME minimum size should be 98304 bytes (0x18000 bytes)
Truncating file at 0x18000...
Checking the FTPR RSA signature... VALID
Done! Good luck!
```
Good news:
```
user@heads-x230:~$ ls -al heads/blobs/x230/me.bin heads/blobs/x230/extracted_me.bin 
-rw-r--r-- 1 user user 98304 Mar 15 09:48 heads/blobs/x230/extracted_me.bin
-rw-rw-r-- 1 user user 98304 Feb 17 12:17 heads/blobs/x230/me.bin
```

Weird:

```
user@heads-x230:~$ sha256sum heads/blobs/x230/me.bin heads/blobs/x230/extracted_me.bin 
079b38c2ffaee7b0d170a10fafd802f25682bf8f0c10941f3744dfec57e6e0dc  heads/blobs/x230/me.bin
c140d04d792bed555e616065d48bdc327bb78f0213ccc54c0ae95f12b28896a4  heads/blobs/x230/extracted_me.bin
```

tlaurion, 2020-03-15:
Here is what I propose. Please see README file. We could implement a module file that would extract me.bin automatically. No way to do the same for ifd.bin for the moment.

https://github.com/osresearch/heads/compare/master...tlaurion:x230_libremkey-repro-gitlab-circleci_blobs?expand=1

@osresearch @merge @kylerankin @MrChromebox @SebastianMcMillan  Thoughts?

MrChromebox, 2020-03-15:
@tlaurion so you're proposing to put the blobs in the main Heads repo, along with reproducibility instructions? Any reason to not put them in a submodule as was discussed previously?

tlaurion, 2020-03-15:
Was a Poc!
Can work around and put them in a submodule.

tlaurion, 2020-03-15:
@MrChromebox : ifd.bin still has to be extracted manually from an already flashed ROM image so solution is not complete as of now.

What I could do is have ifd.bin already there, and me.bin being extracted as a xx30-blobs submodule if specified in board config. That is the way to go?

tlaurion, 2020-03-15:
@MrChromebox : i'm confused. Though you didn't want each build to download from Lenovo [here](https://github.com/osresearch/heads/issues/307#issuecomment-598479213)

MrChromebox, 2020-03-15:
@tlaurion can't the IFD be extracted from the FL1 via g2uj31us.exe ?

tlaurion, 2020-03-15:
@MrChromebox haven't found a way to have a workable 8mb image to apply me_cleaner on, following [precedent ](https://github.com/osresearch/heads/issues/307#issuecomment-599074766) recipe and tickets open at [chipsec](https://github.com/chipsec/chipsec/issues/870) and [uefi-firmware-parser](https://github.com/theopolis/uefi-firmware-parser/issues/77)

n4ru, 2020-03-15:
Extracting from a full ROM is easy, but I thought the point was to get rid of the requirement?

tlaurion, 2020-03-15:
@n4ru : Not extracting from a currently flashed ROM, but extracting from update files (FL1 FL2)! :)

tlaurion, 2020-03-15:
@n4ru: As of right now, me.bin is consistent and [extracted directly from your provided link](https://github.com/osresearch/heads/commit/48267b4b88358ed9bbf6bda28296809ff7b9f77b). That can be scripted. 

But ifd.bin still comes from a [flashed rom](https://github.com/osresearch/heads/commit/fbfe30e78f8c65df6b7d426f265bd89f6dca835f), with unlocked regions and descriptor having AltMeBit set in. Cannot be done by script from original Lenovo provided exe right now.

tlaurion, 2020-03-17:
@MrChromebox how?

MrChromebox, 2020-03-17:
@tlaurion I was under the impression that the FL1 was a full flash image, is it not?

PatrickRudolph, 2020-03-17:
> @tlaurion can't the IFD be extracted from the FL1 via g2uj31us.exe ?

No as it doesn't contain IFD or ME. It looks like an UEFI update capsule.
You can use the UEFITool to open it and see it's contents.

The FL2 seems to be an EC update only.

Maybe other update files contain a ME, not sure about the IFD.

PatrickRudolph, 2020-03-17:
There's no need to use a Lenovo BIOS for this, any BIOS update from any vendor for the same PCH which contains a full ROM will do.

n4ru, 2020-03-17:
> There's no need to use a Lenovo BIOS for this, any BIOS update from any vendor for the same PCH which contains a full ROM will do.

Incorrect, as we've already surmised there are no BIOS updates with a "full ROM".

FL1 contains the BIOS module.
FL2 contains the EC module.
ME installer contains the ME production binary, but not the entire UEFI module.
IFD still missing.

[28 more replies]

PR #424 (2019-02-08): x230: remove 4M and 8M split-images from the build
The bios regions of the 12M coreboot image is 7M: 4M and 3 of the 8M split
image. The rest of the 8M image _generated_ with fake data and not usable
on real systems! It's dangerous to create them and suggest flashing them
externally.

That's exactly why the x230-flash build target is there: To
have a self-contained 4M image and enable easy unlocking of the 8M image
using the _original_ data.

the heads-wiki project is updated accordingly.

Closes #307
Closes #302
commit 4ea17583 (2019-02-08): x230: remove 4M and 8M split-images from the build

The bios regions of the 12M coreboot image is 7M: 4M and 3 of the 8M split
image. The rest of the 8M image _generated_ with fake data and not usable
on real systems! It's dangerous to create them and suggest flashing them
externally.

That's exactly why the x230-flash build target is there: To
have a self-contained 4M image and enable easy unlocking of the 8M image
using the _original_ data.

the heads-wiki project is updated accordingly.

Closes #307
Closes #302
closed 2020-12-02 by tlaurion