linuxboot/heads #351: PCIe bifurcation doesn't work

vejmarie, 2018-03-09:
If a PCIe slot is using bifurcation to implement a dual x8 ports as an example on a x16 port, the attached PCIe device is not recognized. That is an issue on most OCP nodes which are intensively using PCIe bifurcation. A workaround is to directly plugged the PCI card into the master slot.

nospam2000, 2024-05-05:
has there been any progress on this topic?

tlaurion, 2024-05-05:
> There are a couple of registers which could be set at CPU level (not PCH) to set it up. This is definitly something which is transparent to the kernel and can be setup currently through AMITSE screen. I believe we have to re-implement it into the kernel to check if this is PEI or could be performed later.

@vejmarie ping?

I have not followed linuxboot integration for a really long while to be honest and the Heads tree for those boards is totally UNMAINTAINED as of now. If there was desire, we could team up to add supported linuxboot boards back into Heads and build the ROMs from Circleci.

AFAIK, there was no progress on this @nospam2000 .