linuxboot/heads #352: Make fails to build with `make-4.2/glob/glob.c:1367: undefined reference to `__alloca'`

paulmenzel, 2018-03-10:
Make 4.2 fails [to build with a libc 2.27](https://lists.nongnu.org/archive/html/bug-make/2017-11/msg00027.html), shipped now in Debian Sid/unstable.

```
$ make
make  all-recursive
make[1]: Entering directory '/home/joey/src/heads/build/make-4.2'
Making all in glob
make[2]: Entering directory '/home/joey/src/heads/build/make-4.2/glob'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/joey/src/heads/build/make-4.2/glob'
Making all in config
make[2]: Entering directory '/home/joey/src/heads/build/make-4.2/config'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/joey/src/heads/build/make-4.2/config'
Making all in po
make[2]: Entering directory '/home/joey/src/heads/build/make-4.2/po'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/joey/src/heads/build/make-4.2/po'
Making all in doc
make[2]: Entering directory '/home/joey/src/heads/build/make-4.2/doc'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/joey/src/heads/build/make-4.2/doc'
make[2]: Entering directory '/home/joey/src/heads/build/make-4.2'
gcc  -g -O2 -Wl,--export-dynamic  -o make ar.o arscan.o commands.o default.o dir.o expand.o file.o function.o getopt.o getopt1.o guile.o implicit.o job.o load.o loadapi.o main.o misc.o posixos.o output.o read.o remake.o rule.o signame.o strcache.o variable.o version.o vpath.o hash.o remote-stub.o glob/libglob.a   -ldl 
glob/libglob.a(glob.o): In function `glob_in_dir':
/home/joey/src/heads/build/make-4.2/glob/glob.c:1367: undefined reference to `__alloca'
/home/joey/src/heads/build/make-4.2/glob/glob.c:1342: undefined reference to `__alloca'
/home/joey/src/heads/build/make-4.2/glob/glob.c:1283: undefined reference to `__alloca'
/home/joey/src/heads/build/make-4.2/glob/glob.c:1256: undefined reference to `__alloca'
glob/libglob.a(glob.o): In function `glob':
/home/joey/src/heads/build/make-4.2/glob/glob.c:581: undefined reference to `__alloca'
glob/libglob.a(glob.o):/home/joey/src/heads/build/make-4.2/glob/glob.c:732: more undefined references to `__alloca' follow
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:656: make] Error 1
make[2]: Leaving directory '/home/joey/src/heads/build/make-4.2'
make[1]: *** [Makefile:798: all-recursive] Error 1
make[1]: Leaving directory '/home/joey/src/heads/build/make-4.2'
make: *** [Makefile:534: all] Error 2
```

meskio, 2018-03-15:
I'm seeing the same problem in debian sid. Using debian stretch works fine.

osresearch, 2018-09-18:
I'm re-opening this since it seems to still be a problem on ubuntu 18.04

osresearch, 2018-09-18:
This defines `__alloca` so that it works on ubuntu 18.04: https://github.com/osresearch/heads/blob/make-4.2.1/patches/make-4.2.1.patch