linuxboot/heads #353: Do not rebuild Make if Make 4.2.1 is installed

paulmenzel, 2018-03-10:
Currently, the only allowed GNU Make version is 4.2.

```
$ git grep make_version modules/make
modules/make:make_version := 4.2
modules/make:make_dir := make-$(make_version)
modules/make:make_tar := make-$(make_version).tar.bz2
```

Version 4.2.1 should also be allowed.

```
$ make --version
GNU Make 4.2.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

whohoho, 2018-03-12:
It would be nice to allow also higher versions of make, make 4.2 and 4.2.1 don't build on debian:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=891365

But the git head does build.

osresearch, 2018-03-14:
Having a consistent version of make is nice, although it does introduce an extra stage and complicates the build targets. We could possibly remove the make version check entirely... it was to support an ancient build machine inside a corporate firewall that was on make 3.x and didn't support some of the fancier options.

osresearch, 2018-03-14:
We should definitely update where we fetch make from... This doesn't look legit:
```
make_url := http://gnu.mirror.constant.com/make/$(make_tar)
```