linuxboot/heads #356: Impose Access Restriction to the Recovery Shell

biergaizi, 2018-03-11:
Currently, HEADS will prompt a recovery shell under any fault conditions, it may not be a concern in a perspective of security. However, unlike other bootloaders, such as GRUB which only has read-only access to the disk with minimum risks of data destruction, HEADS is a fully-functional Linux-based operating system with full R/W access to the hard drive and hardware.

It enables extremely easy vandalism for script kiddies with physical access to the machine via the recovery shell, i.e. a keyboard-only denial-of-service attack.  It may be a serious issue depending on one's threat model. For example, public computers are especially vulnerable. For example, major bootloaders, such an GRUB, allows programming a hashes password to the configuration script for access restrictions to the boot entry and the shell prompt. With access control, vandalism is only possible by physically disassemble the machine.

In order to implement access control, two types of changes are needed. 

1. First, the recovery shell is provided unconditionally under a fault condition, for example, in `initrd/init`.

    ```
    # We should never reach here, but just in case...
    recovery 'Boot script failure?  Entering recovery shell'
    # belts and suspenders, just in case...
    tpm extend -ix 4 -ic recovery
    exec /bin/ash
    ```

    `exec /bin/ash` should be replaced with `exit` to trigger a kernel panic immediately.

2. `initrd/bin/generic-init` should include an authentication scheme, for example a hashed password checkpoint, or a digital signature verification routine, to ensure the recovery shell access is obtained only under proper authentication.

biergaizi, 2018-03-11:
3. Is it possible to bypass the checkpoints by pressing Control-C? If so, it can be pretty problematic.

flammit, 2018-03-11:
Adding #217 to this thread and closing it as a dup.

persmule, 2018-03-12:
Unexpected fault conditions should be handled with die() rather than recovery().
@citypw Could you help us to review resolutions of this issue?

itay-grudev, 2018-11-29:
This will be relatively easy to bypass. You boot from USB and re-flash the BIOS with a factory default. Essentially this only protects from an inexperienced evil maid tampering with the recovery shell (Not so inexperienced if they can know how to use the recovery shell).

To make this better you will need to password protect boot options as well. And then you threat is: an evil maid with an SPI programmer. 

In the end I am not sure this is going to be very beneficial. If you forget your password you will have to bother with the steps above to defeat this benign "security" measure.

Of course resetting the boot loader is tamper evident. But so is modifying anything - the boot loader or the `/boot` partition. So the overall security improvement of this is essentially minimal. Wouldn't you agree?

biergaizi, 2018-11-29:
> And then you threat is: an evil maid with an SPI programmer.

No. An individual under the threat of an evil maid needs to adopt a radically different set of security protections. Instead, the threat model for this is clearly stated in the beginning.

> vandalism from script kiddies with physical access to the machine via the recovery shell, i.e. a keyboard-only denial-of-service attack. With keyboard-only access to an unguarded machine (such as a public machine in a lab),  i.e. a keyboard-only denial-of-service attack. 

It's a fairly realistic scenario. Have you seen what a 14-year-old script kiddie are capable of doing :smile: ?

tlaurion, 2020-03-09:
@biergaizi @citypw @persemule @itay-grudev?

Following you discussion:
- USB Security device could be used as per #361 on recovery shell and usb code path if defined in board config.

Would that deal with all your concerns? Problem here would be to have a way to put back private keys inside of new USB security dongle if lost, which would require in recovery keypair generation, saved on encrypted external storage and then those keys to be copied in USB security dongle so that it is possible to copy them back on USB security dongle when needed, while permitting users to boot unsigned boot configurations that are already on disk while unsigned per user. I have not better idea.

tlaurion, 2020-08-01:
Still interesting

itay-grudev, 2020-08-03:
@tlaurion I believe your suggestion would stop the specified by @biergaizi script kiddie attack in the ever so elegant way using Librem keys, so I agree - this solution does make sense,

tlaurion, 2021-01-25:
Autnehtication should also be asked prior of flashing ROM.

Not sure where/how to put this correctly in line with #771, which needs to reflash rom with public key generated, which would require flash authentication. (With a private key for which there is no public key but in local keychain. Check against local keychain?) 

@flammit : ideas?

tlaurion, 2022-01-24:
> Authentication should also be asked prior of flashing ROM. (prior of all flashrom calls)
> 
> Not sure where/how to put this correctly in line with #771, which needs to reflash rom with public key generated, which would require flash authentication. (With a private key for which there is no public key but in local keychain. Check against local keychain?)

Actually, as discussed https://github.com/osresearch/heads/pull/326#issuecomment-1019684512 there might not be any need to authenticate flashing operations, which could be the mitigation plan in case of having lost USB Security dongle being replaced by a new one, or having an expired public key needing to be extended. 

Factory resetting newly received USB Security dongle replacement/Flashing a new public key would resolve the issue there, while what is really needed is a way to authenticate at the recovery shell. As in referred link above, the two thought authentication methods would be: 
- TPM Disk Unlock Key passphrase, validating that all firmware measurements are valid and sealed with LUKS now measured full headers and TPM NVRAM region passphrase to unseal the secret.
- GPG User PIN to test signing capability of the user against public key already fused inside of ROM, confirming both PIN and public key fused are valid.

Thoughts?

tlaurion, 2023-11-28:
Integrated in #1515

closed 2023-11-28 by tlaurion