linuxboot/heads #372: Dual boot and NVMe support

kakaroto, 2018-03-29:
NVMe drives are not currently supported in Heads. Although you could set the boot device in the config to /dev/nvme0n1p1 and have NVMe support in the kernel and consider it enough, there are a couple of issues here.

* First, booting into an NVMe drive doesn't actually work. After it prints "Executing kernel" (or whatever it prints when doing the kexec), the computer simply reboots into Heads
  * I also had the same behavior when booting into a USB drive (a full install in a drive in a USB enclosure, not a live usb)
* We expect a single heads image to be deployed for all customers and that same image needs to handle the various use cases : 
  * a SATA drive, 
  * an NVMe drive
  * a SATA drive *and* an NVMe drive
  * any of the above + a USB device
* What if a user has two drives with both drives having a system installed (let's say Qubes in one and PureOS on the other) and want to dual boot them.

First problem for NVMe support is going to be figuring out why the machine reboots when doing the kexec off the NVMe drive (this can be filed as a separate issue though since it also affects USB booting), but the main work to be done here is in how to handle the boot devices. Currently the boot device is set as a CONFIG_BOOT_DEV in the board config file, but if it's set to /dev/sda1 (which gets it added to fstab), and you have NVMe, it will fail, if you have the OS in NVMe but you also have /dev/sda1, it will mount it but fail to parse /boot, if you have an NVMe drive but also a USB device, then /dev/sda1 that will get mounted will be the USB drive (assuming usb modules already got loaded). This of course also affects the hardcoded CONFIG_USB_BOOT_DEV.
And finally, like I said, if you have two (or more) drives, I would expect the script to be able to detect them and offer the choice "Default Boot: Evo 960 NVMe" and "Default Boot: Seagate ST3000", etc...

osresearch, 2018-03-29:
On Winterfell/Leopard/TiogaPass servers with Heads+LinuxBoot we were able to boot from NVMe without any problems related to the drive.  I wonder what is going wrong with the coreboot path.

vejmarie, 2018-03-29:
sorry to ask the question, but what is the underlying hardware ? (aka server model, and nvme drive model). Can you post the kexec command you issue and the associated parameters ? The reboot is happening straight after kexec ? Or you have to wait a few seconds.

kakaroto, 2018-03-29:
@vejmarie it's on Purism Librem 13 v2 and Librem 13 v3. I tried two different machines, same behavior, the NVMe drive is a Samsung Evo 960. I don't know what the USB enclosure was (it was from someone else who said "try this"). the kexec command is just  "root=/dev/mapper/crypt-root ro quiet splash", heads itself will also add "intel_iommu=on intel_iommu=igfx_off", and after the kexec, the screen will blank, and after a few seconds I see heads booting. Note though that it takes a couple of seconds for coreboot to load heads, so I don't know if it reboots right away and the screen being blank is just waiting for heads to be loaded, or if it boots the new kernel (but nothing is displayed due to quiet flag) and then that causes the reboot. I can test more next week if you need additional information.

itay-grudev, 2018-12-02:
I don't know if this hasn't been fixed somehow, but I am running a Librem 13 v3 with an Intel NVMe 760p SSD. I'm booting from `/dev/nvme0n1p1` without a problem.

tlaurion, 2020-03-09:
@MrChromebox ?

MrChromebox, 2020-03-09:
@tlaurion NVMe drives work without issue across all Librem drives. How we want to handle dual boot from 2 separate /boot partitions, regardless of whether they exist on the same or different physical drives, is an entirely different question

closed 2020-03-09 by tlaurion

FrostKnight, 2021-02-23:
Is there plans to support multi-boot/multi FDE + /boot in the future?

Meaning multi full disk encryption or dual booting in general?