linuxboot/heads #375: X230-flash image is 256Kb, not 4096Kb

tlaurion, 2018-03-31:
```
user@build-x230:~/Downloads/heads-x230t$ ls build/x230-flash/coreboot.rom -alh
-rw-r--r-- 1 user user 256K Mar 31 13:22 build/x230-flash/coreboot.rom
```

tlaurion, 2018-03-31:
https://github.com/osresearch/heads/blob/master/boards/x230-flash/x230-flash.config#L12
No coreboot file file?!

osresearch, 2018-04-02:
The default config name is `config/coreboot-$(BOARD).config`, but the board is named `x230-flash` and the file is `coreboot-x230.flash.config`.  Should be an easy fix to rename it -- I'm building a test now.

tlaurion, 2018-04-02:
Resulting size was the same for me applying that change manually. It's
fixed?

osresearch, 2018-04-02:
Works for me on a clean build:
```
~/heads: ls -Fla build/x230-flash/coreboot.rom 
-rw-r--r-- 1 user user 12582912 Apr  2 11:10 build/x230-flash/coreboot.rom
```

Your coreboot was probably not rebuilt -- the `config/coreboot-x230-flash.config` is likely older than the `build/coreboot-4.7/.configured`, so Make didn't think it had to rebuild.

(And that looks like another latent bug: it should probably depend on `build/coreboot-4.7/x230-flash/.configured`)

tlaurion, 2018-04-11:
@osresearch :12582912 is 12M, not 4M! :)

osresearch, 2018-04-11:
Oh, right...  There must have been a make rule at some point that used `dd` to extract the top 4 MB of that 12 MB image that was lost.

This sort of specialization argues for us to add special rules into the `boards/x230-flash/x230-flash.config` and recognize that those are actually Makefiles.

tlaurion, 2018-04-17:
@osresearch : please reopen. I have no idea how to implement this other then in the Makefile directly like it was done before.

osresearch, 2018-04-18:
Reopening -- we need to add the makefile rule to extract just the top 4M from the 12 MB image.

osresearch, 2018-04-18:
Fixed both x230 and x230-flash.

tlaurion, 2018-04-19:
@osresearch: A clean build gives:
"*** No rule to make target '/home/user/Downloads/heads/build/x230/coreboot.rom', needed by '/home/user/Downloads/heads/build/x230/x230-8.rom'.  Stop."

osresearch, 2018-04-19:
works with make-4.2.1, but not system make... 

oh, `boards/$(BOARD)/$(BOARD).config` is included before the make check.  Re-opening....

tlaurion, 2018-04-19:
recipe for target '/home/user/Downloads/heads/build/make-4.2/Makefile' failed
make: *** [/home/user/Downloads/heads/build/make-4.2/Makefile] Error 1

This is my quick clean test i made to replicate:
```
tar zcvf cross_gcc47.gz ./coreboot-4.7/util/crossgcc/tarballs/
rm build/* -rf && tar zxvf coreboot_47.tar.gz && make BOARD=x230-flash
```

tlaurion, 2018-04-19:
I still think the best would be to have a 4MB (x230-flash) and a 12MB image (x230), since internal flashing is always done internally *BUT* when screwing up things in testing.

tlaurion, 2018-04-20:
Currently, three images are generated. I think it fits all needs.
```
2018-04-20 14:03:40-04:00 EXTRACT build/x230/coreboot.rom
30616a7b3e290b355c0c703b8484bbcf2220e14babbe5f8008c1534c6e56ae10  build/x230/coreboot.rom
dd of=/home/user/Downloads/heads-x230t/build/x230/x230-8.rom if=/home/user/Downloads/heads-x230t/build/x230/coreboot.rom bs=65536 count=128 skip=0
128+0 records in
128+0 records out
8388608 bytes (8.4 MB, 8.0 MiB) copied, 0.0142396 s, 589 MB/s
sha256sum /home/user/Downloads/heads-x230t/build/x230/x230-8.rom
bd8c640970a9bb10878d9daa76d704edb2fa5fe468e51da53e2588f380c84ef6  /home/user/Downloads/heads-x230t/build/x230/x230-8.rom
dd of=/home/user/Downloads/heads-x230t/build/x230/x230-4.rom if=/home/user/Downloads/heads-x230t/build/x230/coreboot.rom bs=65536 count=64 skip=128
64+0 records in
64+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.00318218 s, 1.3 GB/s
sha256sum /home/user/Downloads/heads-x230t/build/x230/x230-4.rom
bf00e8b86201ad18db47b14c596d5028727e4f034128bca41b0681d33a278b9e  /home/user/Downloads/heads-x230t/build/x230/x230-4.rom
make[1]: Leaving directory '/home/user/Downloads/heads-x230t'
```

tlaurion, 2018-04-20:
Flashing the 4mb results on screen:
"XZ-Compressed data is corrupt"

Seems like the rom is cut in half...

tlaurion, 2018-04-20:
I confirm:
user@build-x230:~/Downloads/heads-x230t/build/coreboot-4.7/util/cbfstool$ ./cbfstool ../../../../build/x230/x230-8.rom print
E: FMAP records image size as 12582912, but file is only 8388608 bytes (did something truncate this file?)
user@build-x230:~/Downloads/heads-x230t/build/coreboot-4.7/util/cbfstool$ ./cbfstool ../../../../build/x230/x230.rom print
../../../../build/x230/x230.rom: No such file or directory
user@build-x230:~/Downloads/heads-x230t/build/coreboot-4.7/util/cbfstool$ ./cbfstool ../../../../build/x230/coreboot.rom print
Name                           Offset     Type           Size   Comp
cbfs master header             0x0        cbfs header        32 none
fallback/romstage              0x80       stage           80132 none
cpu_microcode_blob.bin         0x13a00    microcode       22528 none
cmos_layout.bin                0x19280    cmos_layout      1804 none
fallback/dsdt.aml              0x19a00    raw             13390 none
(empty)                        0x1cec0    null            12248 none
mrc.cache                      0x1fec0    mrc_cache       65536 none
fallback/ramstage              0x2ff00    stage           89793 none
fallback/payload               0x45e00    payload       4928095 none
(empty)                        0x4f90c0   null          2123288 none
bootblock                      0x6ff700   bootblock        1968 none

osresearch, 2018-04-20:
The `x230-4.rom` and `x230-8.rom` must be both flashed, one onto each chip.  They are not standalone.

The `x230-flash.rom` is 4MB and should be self contained.

tlaurion, 2018-04-20:
Makes sense. x230-flash works.
Flashing x230-4mb and x230-8mb doesn't, since it overwrites ME and makes the system unbootable.

@osresearch What is the utility of seperated 4bm and 8mb files?