linuxboot/heads #376: Support for a more flexible GPG auth model

flammit, 2018-03-31:
Currently there is a single gpg keyring that is used to determine which keys are trusted in a Heads installation.  Unfortunately, there are now different ways `gpgv` is used in `init`:

1) Verify when booting from local disks that the kexec config valid
2) Verify when booting from an ISO that is valid

In the future, there will probably be additional uses:

3) Validate access to the recovery shell (as in https://github.com/osresearch/heads/pull/361)
4) Validate normal boot procedure with 2FA
5) Decrypt the TPM-sealed luks disk key (instead of user-provided password)

It seems like maintaining a few key rings would be sufficient to cover a wide range of cases:
`/.gnupg/user/`: valid sigs to do anything in Heads
`/.gnupg/admin/`: additional valid sigs to access a recovery shell to wipe/re-provision hardware (but not access user's key material)
`/.gnupg/distro/`: additional valid sigs to ISO boot

On a standard reproducible build, Heads would have empty `user` and `admin` ring, and the `distro` ring would contain whatever standard distro keys are available (e.g. qubes, tails, alpine, etc.).  An installer could then use `cbfstool` / https://github.com/osresearch/heads/pull/357 to add a public key into `admin` or necessary public key and private key stubs to `user`.

@osresearch @tlaurion @zaolin @kakaroto @kylerankin thoughts?

tlaurion, 2018-03-31:
I love it. I think it would do the job entirely and fit all use cases.

kakaroto, 2018-04-30:
I think it's a great idea as well. I vote yes.

tlaurion, 2018-05-05:
Key installation doc [here](https://github.com/osresearch/heads/pull/357#issuecomment-383236197)

tlaurion, 2018-08-03:
@flammit This was merged!

tlaurion, 2019-01-26:
So.

This [section is useless in GPG2](https://github.com/osresearch/heads/pull/510/files#diff-49199d392e898d67aa3a302abee66c19R147)

@flammit : 
When using Heads with Whiptail, the user can upgrade the rom with two options: 
1. With a "clean rom": a clean rom won't contain any user public key. When rebooting, @kylerankin's #490 will trigger and ask the user if he wants to import a new key. On top of this, [with this PR](https://github.com/kylerankin/heads/pull/2), the user is provided with the option to Factory reset a Nitrokey/LibremKey, generate keys, export public key and trusdb, and insert those into a taken running rom export prior to reflashing it back internally into SPI flash.
2. with a new rom: injects current user public key and otrust.txt into rom prior to flashing it into SPI flash.

closed 2020-03-11 by tlaurion