linuxboot/heads #381: TPM reset command should be followed by removing /boot/kexec*

tlaurion, 2018-04-08:
https://github.com/osresearch/heads/blob/f60f0543d633651c3df0742c6e9e17a27149b365/initrd/bin/gui-init#L127

Else, following boot will fail.

tlaurion, 2018-04-09:
@kakaroto  @flammit @osresearch 
What do you think?

kakaroto, 2018-04-09:
Not kexec-*, just kexec-rollback.txt actually since it contains the counter id of the TPM. This problem also occurs if you swap your HDD from one machine to another since the TPM will be different and the counter id will not match either. This is a known issue that I reported to @kylerankin a few weeks ago and it's in his TODO list of use cases to fix for 'common users'.

flammit, 2018-04-10:
The UX around TPM counter / rollback management is difficult because those counter files might exist on devices that are not even connected at the time of TPM reset.  Easily resetting the rollback protection is difficult because of the need to differentiate attack scenarios from the user-driven ones.

tlaurion, 2018-04-10:
So maybe the solution lies should be implemented where rollback checks are
done. Maybe ask for gpg dongle to permit deletion?

kakaroto, 2018-04-11:
I'm not entirely sure I fully understand the rollback/counter protections and attacks. By the way, yeah, if you delete the rollback file, the signatures will fail, so you might as well delete all of the kexec-* files.
@tlaurion I don't think you should need the gpg key to permit deletion because anyone could delete the files manually from the OS (or by swapping HDD and using another computer to mount the partition). So I think deleting the files is not an issue as long as the user will see a "the files are not signed" the next time they try to boot and they already need the gpg key to sign the files.

tlaurion, 2018-04-17:
> The UX around TPM counter / rollback management is difficult because those counter files might exist on devices that are not even connected at the time of TPM reset. Easily resetting the rollback [...]

@flammit :
In the use case i'm enforcing, I want the user to be able to boot into a working build environment, deployed as a seperate build-template and working build environment so he can integrate his own keys in the rom and reflash that image. From that point, the user will take ownership of the system, reset the disk recovery password with his own and change user password inside of Qubes, or wipe the system completely if he wishes.

The point is that the user will need to take ownership of the machine. In any case, a TPM reset should delete kexec* files if present. Resetting the TPM itself should ask for the GPG dongle in my opinion, like any other change done under Heads.

Other opinions?

flammit, 2018-04-29:
As discussed on slack w/ @osresearch, I think a better design than deleting these files would be to detect bad rollback situations on boot attempts, then allow the user to reset the counter and resign the configs once they've confirmed that it is an expected situation.  That way it will work for all manner of rollback failure (`tpm-reset`, old external boot drives, etc.).

As an additional feature, once we have #357 merged along with a working version of `poke`ing to update small portions of the internal ROM, we should move the boot location-specific `kexec-rollback` files to a single file ROM and then allow multiple boot configs to reference a single good rollback checkpoint (i.e. loading that ROM file into something like `/etc/tpm-rollback` and then referencing that in the set of files for `kexec.sig` generation).

closed 2018-11-08 by tlaurion