linuxboot/heads #385: slang fails on clean checkout and maybe odd numberes builds

osresearch, 2018-04-19:
At least, I think that is what is going on.  @kylerankin or @kakaroto?

```
mkdir -p -- /root/x230/install//share/slsh
mkdir: cannot create directory '/root/x230/install//share/slsh': File exists
make[3]: *** [Makefile:121: install_directories] Error 1
```

osresearch, 2018-04-19:
Although I'm confused, since `mkdir` appears to say that is not an error:

```
     -p, --parents
              no error if existing, make parent directories as needed
```

osresearch, 2018-04-19:
And sometimes the error is diffrent: ```
/usr/bin/install: cannot stat '/root/x230/build/slang-2.3.1a/src/elfobjs/libslang.so.2.3.1': No such file or directory
make[3]: *** [Makefile:210: install-elf-and-links] Error 1
make[3]: *** Waiting for unfinished jobs....
```

kylerankin, 2018-04-20:
Not to hijack the bug but I'm having trouble reproducing because at least on my system, I've noticed a long-standing issue with recent Heads builds where I can't get as far as building make:
``glob/libglob.a(glob.o): In function `glob_in_dir':
/home/kyle/heads/build/make-4.2/glob/glob.c:1367: undefined reference to `___alloca'

along with a lot of the same error for other lines in that file. 

I'm not sure whether there is some new library dependency I need to add or what the issue is, but I've seen this on clean test machines and also on the system I typically use to build Heads (and have built Heads on in the past). On the latter system once I copy a pre-existing build/make-4.2 directory over to the clean Heads checkout I can build the rest.

osresearch, 2018-04-20:
You're not the only one with that bug: https://github.com/osresearch/heads/issues/352

osresearch, 2018-04-20:
Based on a suggestion from @flammit I've disabled parallel makes during the `install` target for slang.

closed 2018-04-20 by osresearch