linuxboot/heads #402: GPG should be bundled with x230-flash and x220-flash

tlaurion, 2018-05-06:
This way, a new user could generate his keys directly from within heads and integrate them in the rom to be flashed, without the need of a second computer.

tlaurion, 2018-05-06:
Building to see  if there is enough place for it in the 4mb image.

osresearch, 2018-05-07:
What's the value in having it in the flasher image, versus the full image? Once the user has flashed the system with the full image, they need to reboot to take ownership of the TPM, install their gpg keys, seal the disk keys, etc.

tlaurion, 2018-05-07:
Not flash 3 times.

1- Inside x230-flash
Generate keys to external media.
Export armored keys to external media.
Add armored keys inside of downloaded/built ROM.
Add distribution trusted signatures.
Flash full ROM
2-Inside x230
Take ownership of TPM.
Boot from USB iso. Install OS. Reboot.
Select default boot, sign config, seal disk keys in one step.

Done.

Else, reflashing the ROM invalidates measurements and configs. Two steps is
more straightforward then 3 steps, even logically, no?

merge, 2018-05-25:
I for one would question the x230-flash config in general, and the 4m/8m split. It suggests flashing the 8m image which is not usable.

I'd document flashing _any_ 4m-coreboot-distribution-image and (or only) unlock the IFD on the 8m chip, and moving to a full Heads from there, something like [this](https://github.com/merge/skulls/blob/master/x230/README.md#moving-to-heads).

Anyways, please never _depend_ on having x230-flash actually flashed!

flammit, 2018-05-25:
@merge agreed that having the 8MB chunk from the full 12MB x230 image is a bit misleading and should probably be removed.  x230-flash isn't required, just usually the fastest if you're externally flashing the first time.

I often use the 4M x230-flash image to recover bricked systems.  It is also the easiest way to configure the Heads runtime would be within the runtime (since then you won't need a compiled cbfstool).

The skulls install docs are quite good, so if you have suggestions for the Heads docs please PR to the heads-wiki =)

tlaurion, 2018-06-07:
See https://github.com/osresearch/heads/pull/404

tlaurion, 2018-06-07:
I agree that x230 board target should only generate coreboot.rom (why not x230.rom btw?), and no 4mb and 8mb files. (I still don't quite understand their existence presence.)

That 12mb is to be flashed from within heads, preferably with user keys bundled in.

Didn't try to fit whiptail (not FB) in x230-flash.rom  and replace the board init script with gui-flash.sh, but if that fits, the UX would be nearly flawless IMHO. The user could then take x230.rom, generate his keys and add them into the rom in one, easy step.

closed 2018-07-18 by osresearch