linuxboot/heads #416: Use minimal configurations for Kconfig configuration files (Linux, coreboot)

paulmenzel, 2018-05-19:
In projects using Kconfig, `make help` shows the description below for *savedefconfig*.

> savedefconfig   - Save current config as ./defconfig (minimal config)

Using the minimal configuration avoids the hassle to update configuration files on version updates even when nothing from our selections actually changed. It’s also easier to grasp, what is actually selected as it’s much shorter.

paulmenzel, 2018-07-18:
On 07/18/18 13:28, Trammell Hudson wrote:
> For `qemu-linuxboot` the generated `defconfig` file is quite a bit
> shorter (9K versus 16KB).
> 
> How do you apply it?  Running `make defconfig` uses
> `x86_64_defconfig`, although I'm not sure where it is coming from.

I copy `defconfig` to `.config`, and then do `make olddefconfig`.

osresearch, 2018-08-09:
This is now implemented in the 4.14 update PR: https://github.com/osresearch/heads/commit/f4e25dd2168ac414398bbc9e57bb5c7138432173

It cleans up the `config/linux-*.config` files immensely.  Thanks for suggesting it.

closed 2018-08-09 by osresearch