linuxboot/heads #423: Update to Linux 4.14 with CBMEM console support

paulmenzel, 2018-05-27:
Since Linux 4.12, selecting `GOOGLE_MEMCONSOLE_COREBOOT` exports the CBMEM console under `/sys/firmware/log`, cf. commit d384d6f4 (firmware: google memconsole: Add coreboot support).

To not rely on `cbmem`, it’d be a good idea to update to Linux 4.14, as that is the latest LTS series.

osresearch, 2018-07-18:
I've built a branch with 4.14.56 support: https://github.com/osresearch/heads/tree/linux-4.14-update

However with `BOARD=qemu-coreboot` and the current coreboot-4.7 there is no log to be found in `/sys/firmware/log`.

paulmenzel, 2018-07-18:
On 07/18/18 12:21, Trammell Hudson wrote:
> I've built a branch with 4.14.56 support:
> https://github.com/osresearch/heads/tree/linux-4.14-update
> 
> However with `BOARD=qemu-coreboot` and the current coreboot-4.7 there
> is no log to be found in `/sys/firmware/log`.

Did you load the module with `sudo modprobe memconsole_coreboot`?

osresearch, 2018-07-18:
It should be built in, not as a module: https://github.com/osresearch/heads/blob/linux-4.14-update/config/linux-qemu.config#L2166

paulmenzel, 2018-07-18:
On 07/18/18 12:38, Trammell Hudson wrote:
> It should be built in, not as a module:
> https://github.com/osresearch/heads/blob/linux-4.14-update/config/linux-qemu.config#L2166
Understood.

Just to be sure, does `cbmem -c` show the messages?

osresearch, 2018-07-18:
Yes, `cbmem --console` has the output.  I increased the log level to be sure that there would be something on the console: https://github.com/osresearch/heads/commit/a75d8a0c92a331f708d45446b577d06c27a05247

osresearch, 2018-08-13:
4.14 is now in master

closed 2018-08-13 by osresearch