linuxboot/heads #435: Upgrade GnuPG to 2.2.9

tlaurion, 2018-08-03:
Attempting to add dependencies to Heads.

```
configure:
***
*** You need libgpg-error to build this program.
**  This library is for example available at
***   https://gnupg.org/ftp/gcrypt/libgpg-error
*** (at least version 1.24 is required.)
***
configure:
***
*** You need libgcrypt to build this program.
**  This library is for example available at
***   https://gnupg.org/ftp/gcrypt/libgcrypt/
*** (at least version 1.7.0 (API 1) is required.)
***
configure:
***
*** You need libassuan to build this program.
*** This library is for example available at
***   https://gnupg.org/ftp/gcrypt/libassuan/
*** (at least version 2.5.0 (API 2) is required).
***
configure:
***
*** You need libksba to build this program.
*** This library is for example available at
***   https://gnupg.org/ftp/gcrypt/libksba/
*** (at least version 1.3.4 using API 1 is required).
***
configure:
***
*** It is now required to build with support for the
*** New Portable Threads Library (nPth). Please install this
*** library first.  The library is for example available at
***   https://gnupg.org/ftp/gcrypt/npth/
*** (at least version 1.2 (API 1) is required).
***
configure: error: 
***
*** Required libraries not found. Please consult the above messages
*** and install them before running configure again.

```

tlaurion, 2018-08-03:
@duncanguthrie : Do you have `modules/*` dependencies files to share in regard of your https://github.com/osresearch/heads/issues/296#issuecomment-355665121 comment?

duncanguthrie, 2018-08-03:
I made a fork that compiles (but didn't work properly as described in the comment above, which I thought was an issue with cross-compilation), you could use it as a starting point:

https://github.com/duncanguthrie/heads/tree/gnupg2/modules

It packaged all the gpg2 dependencies at the time. I didn't update it since then.

tlaurion, 2018-09-03:
@osresearch if you have time to look at it, there is a crosscompiling issue I can't figure out how to fix.

tlaurion, 2018-09-04:
@duncanguthrie yes.
[image]

Here is the [branch](https://github.com/tlaurion/heads/tree/x230-current_coreboot48_gpg2) tested on a x230.

tlaurion, 2018-09-13:
@duncanguthrie : Have any insight?

tlaurion, 2018-09-17:
@osresearch ?

osresearch, 2018-09-18:
`gpgv` is just a shell script that runs `gpg`:

```
#!/bin/sh
# if we are using the full GPG we need a wrapper for the gpgv executable
exec gpg --verify "$@"
```

I had missed your very subtle "it" link to this tree in a previous comment and am looking at it now: https://github.com/duncanguthrie/heads/tree/gnupg2/modules

osresearch, 2018-09-18:
At a first glance, it looks like issue #313 might be a problem -- if a directory is not listed in `BIN_MODULES` it won't have any executables copied into the initrd. We really should remove that.

I'm also getting a missing `libusb` error:

```
configure:
***
*** You need libusb to build the internal ccid driver.  Please
*** install a libusb suitable for your system.
***
configure: error: 
***
*** Required libraries not found. Please consult the above messages
*** and install them before running configure again.
***
```

osresearch, 2018-09-18:
Looks like it is including `<libusb.h>`, but our libusb installed into `include/libusb-1.0/libusb.h`. Adding `CPPFLAGS="-I$(INSTALL)/include/libusb-1.0` seems to have allowed it to build.

osresearch, 2018-09-18:
My gpg2 branch builds now, based on a pull from @duncanguthrie's gpg2 tree and a few hacks: https://github.com/osresearch/heads/tree/gpg2

In addition to fixing the `BIN_MODULES` flag, I also had to change the libraries that are copied into the intird to include the version numbers.  With those changes:

```
!!!!! Starting recovery shell
~ # gpg --version
gpg (GnuPG) 2.2.4
libgcrypt 1.8.2
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Home: //.gnupg
Supported algorithms:
Pubkey: RSA, ELG, DSA, ECDH, ECDSA, EDDSA
Cipher: IDEA, 3DES, CAST5, BLOWFISH, AES, AES192, AES256, TWOFISH,
        CAMELLIA128, CAMELLIA192, CAMELLIA256
Hash: SHA1, RIPEMD160, SHA256, SHA384, SHA512, SHA224
Compression: Uncompressed, ZIP, ZLIB
~ # 
```

tlaurion, 2018-09-19:
@osresearch @duncanguthrie The current gpg2 branch doesn't include gpg-agent nor scdaemon. Having an agent is required since GPG 2.1. The problem i'm still facing and that is still unresolved with the proposed patch on top of your branch is:
> can't create socket: Address family not supported by protocol.
[gpg2_diff.txt](https://github.com/osresearch/heads/files/2395359/gpg2_diff.txt)

tlaurion, 2018-09-19:
After fixing [this](https://github.com/osresearch/heads/pull/454#discussion_r218914515)

gpg --card-edit --debug-all gives:
[image]

gpg-agent --daemon --debug-all gives:
[image]

scdaemon --daemon --debug-all gives:
[image]

tlaurion, 2018-09-19:
@osresearch : After fixing [this](https://github.com/osresearch/heads/pull/454#discussion_r218914515)

gpg --card-edit --debug-all gives:
[image]

gpg-agent --daemon --debug-all gives:
[image]

scdaemon --daemon --debug-all gives:
[image]

tlaurion, 2018-10-05:
@osresearch  @flammit : fixed under https://github.com/osresearch/heads/pull/462

tlaurion, 2018-10-17:
Actual stopper in [pinentry not being called correctly](https://github.com/osresearch/heads/pull/462#issuecomment-428186021). Any help welcomed.

tlaurion, 2018-11-10:
@osresearch : works [here](https://github.com/tlaurion/heads/tree/x230-current-working-FBWhiptail_required_changes-gpg2)

closed 2019-04-27 by tlaurion