linuxboot/heads #436: Include hdparm

asvc, 2018-08-09:
Might be a silly suggestion, but is it possible to include hdparm and then use it within GRUB to unlock SED drive?

osresearch, 2018-08-09:
Is there a source control tree somewhere? I can't find it other than release tar files on sourceforge? https://sourceforge.net/projects/hdparm/files/latest/download

asvc, 2018-08-11:
SC tree is available from Debian maintainer(s): https://salsa.debian.org/debian/hdparm

tlaurion, 2019-05-01:
@asvc @osresearch : hdparm is a better way to wipe content, as per [ATA Secure Erase function](https://ata.wiki.kernel.org/index.php/ATA_Secure_Erase) then #441. What you think?

asvc, 2019-06-02:
I agree. It also supports `--security-erase-enhanced` which is:

> Enhanced secure erase writes predetermined data patterns (set by the manufacturer) to all user data areas, including sectors that are no longer in use due to reallocation. [URL](https://web.archive.org/web/20110222015452/http://cmrr.ucsd.edu/people/Hughes/HDDEraseReadMe.txt)

@merge , is this something you might consider usable for https://github.com/merge/skulls/issues/45?

tlaurion, 2024-04-16:
Cryptsetup 2.7.1 could do that without adding yet another tool under heads.

closed 2024-04-16 by tlaurion