linuxboot/heads #438: Other suggestions

PowerPress, 2018-08-10:
Is there anything else that can done hardware wise then loading Heads on 230 and using Qubes 4.0+ OS? This technology is great just wandering if there is anything else out there that will compliment it or improve security posture?

Is the Lenovo Thinkpad 230 the best laptop for security features with Heads?

osresearch, 2018-08-13:
There are some steps that are not part of the stock install that we should considering implementing or documenting better:
* Setting the SPI flash non-volatile block protection bits (BP)
* Grounding the write protect pin (WP#) to enforce hardware protection
* Disabling DMA capabilities of the card interface and docking connector
* Using the me_cleaner to reduce the Mangement Engine attack surface
* Protecting the S3 resume script
* Flushing disk encryption keys during sleep
* Measurements when awaking from sleep
* SMM block encryption
* SMM integrity monitoring
* EC firmware (lots of interesting things to investigate)
* Battery firmware
* And probably lots more

Most of these are addressing increasingly esoteric threats, so they may not apply to your use case.

osresearch, 2018-08-13:
On the negative side, the x230 is lacking some number of features that would improve its security:
* Tamper switches (like the X1 or T470)
* Bootguard Measured Boot mode (like the Librem)
* Open source EC (like chromebooks)

PowerPress, 2018-08-13:
Is there another recommended model that will work with Heads and Qubes 4
and still be able to take advantage of these?