linuxboot/heads #441: Implement LUKS keys wipe board option?

tlaurion, 2018-08-24:
In some cases, people may be forced to type in their disk decryption passphrase to boot their laptop in certain unfortunate situations.
 
In those circumstances, those users might desire that their laptop denies further attempts and being themselves unable to use their laptop and to loose access to it then to give away their secrets and disk content.

Wiping LUKS keys after too many failing attempts might be a good solution in this scenario.
A BOARD option permitting to wipe the LUKS headers would permit to enforce such feature and return back to the boot selection menu.

PowerPress, 2018-09-21:
this is a good idea

tlaurion, 2018-10-06:
could be simply implemented by calling `cryptsetup luksKillSlot /dev/sda2 X`, where X is normally 0 and 1, 0 being the recovery disk key and 1, the one released by the TPM under HEADS.

jfrederickson, 2018-10-12:
A more permanent solution could be to wipe/reconfigure the TPM itself. The LUKS headers are stored on disk and could be cloned by the attacker before forcing you to enter the passphrase; it's easy to see that the LUKS headers are missing when comparing the two. In theory the TPM isn't clonable non-destructively.

As far as how to trigger it... I haven't gotten my Heads install up and running quite yet, but I believe if you're using a TPM-sealed key you're prompted for a passphrase? A duress passphrase entered at that prompt would probably be the easiest to conceal, provided the output after using it looks identical to an incorrect entry.

tlaurion, 2018-10-12:
@jfrederickson : There is an option to [use the TPM to seal and decrypt the LUKS volume](https://github.com/osresearch/heads/blob/584c07042ef4898de52ba4981997211c646b75fd/initrd/bin/kexec-save-key) with a key (slot1) that is not the main decryption key (slot 0). That additional key slot is enforced when the user sets a default boot option and decides to set a second decyption password, bound to the integrity attestation from measured boot, which makes it usable from the same computer it was setup on. 

The Idea behind the wipe option is to wipe all the the slots, leaving the disk useless until a LUKS Header backup is restored, only if such backup exists, which comes with it's own security risks. 

I do not think of any other option, and I haven't tested if SSD puts those blocks into erased pool blocks. Unfortunately, the only other option i'm thinking of would be to use hdparm and instruct the disk to secure erase it, [but is manufacturer dependent. Here for Intel firmware](https://www.thomas-krenn.com/en/wiki/SSD_Secure_Erase)

tlaurion, 2019-02-20:
[Interesting](https://www.kali.org/tutorials/emergency-self-destruction-luks-kali/). That would do it.

Todo: 

- [ ] Update [luksAddNuke patch ](https://github.com/offensive-security/cryptsetup-nuke-keys)to cryptsetup 1.7.3
- [ ] Implement LUKS Header backup and restore options in gui-init
- [ ] Implement LUKS luksAddNuke menu in gui-init

1989gironimo, 2019-02-21:
The new repo of this patch/modified cryptsetup is here http://git.kali.org/gitweb/?p=packages/cryptsetup.git;a=summary

tlaurion, 2019-04-16:
For the record. [src](https://gitlab.com/cryptsetup/cryptsetup/wikis/FrequentlyAskedQuestions)
"5.21 Why is there no "Nuke-Option"?

A "Nuke-Option" or "Kill-switch" is a password that when entered upon
unlocking instead wipes the header and all passwords.  So when
somebody forces you to enter your password, you can destroy the data
instead.
While this sounds attractive at first glance, it does not make sense
once a real security analysis is done.  One problem is that you have
to have some kind of HSM (Hardware Security Module) in order to
implement it securely.  In the movies, a HSM starts to smoke and melt
once the Nuke-Option has been activated.  In reality, it just wipes
some battery-backed RAM cells.  A proper HSM costs something like
20'000...100'000 EUR/USD and there a Nuke-Option may make some sense.
BTW, a chipcard or a TPM is not a HSM, although some vendors are
promoting that myth.
Now, a proper HSMs will have a wipe option but not a Nuke-Option,
i.e.  you can explicitly wipe the HSM, but by a different process
than unlocking it takes.  Why is that?  Simple: If somebody can force
you to reveal passwords, then they can also do bad things to you if
you do not or if you enter a nuke password instead.  Think locking
you up for a few years for "destroying evidence" or for far longer
and without trial for being a "terrorist suspect".  No HSM maker will
want to expose its customers to that risk.
Now think of the typical LUKS application scenario, i.e. disk
encryption.  Usually the ones forcing you to hand over your password
will have access to the disk as well, and, if they have any real
suspicion, they will mirror your disk before entering anything
supplied by you.  This neatly negates any Nuke-Option.  If they have
no suspicion (just harassing people that cross some border for
example), the Nuke-Option would work, but see above about likely
negative consequences and remember that a Nuke-Option may not work
reliably on SSD and hybrid drives anyways.
Hence my advice is to never take data that you do not want to reveal
into any such situation in the first place.  There is no need to
transfer data on physical carriers today.  The Internet makes it
quite possible to transfer data between arbitrary places and modern
encryption makes it secure.  If you do it right, nobody will even be
able to identify source or destination.  (How to do that is out of
scope of this document.  It does require advanced skills in this age
of pervasive surveillance.)
Hence, LUKS has not kill option because it would do much more harm
than good.
Still, if you have a good use-case (i.e. non-abstract real-world
situation) where a Nuke-Option would actually be beneficial, please
let me know."

1989gironimo, 2019-04-16:
I think it's useful because you're able to destroy your data really fast. (for instance if the attacker is on the way to you, or in front of your door and you're in a hurry to get rid of of your data) 

It's not suitable for situations where someone's standing behind you with a wrench in their hands and trying to coerce you into revealing your password.

tlaurion, 2019-05-01:
@1989gironimo : Following everything that i've read, i'm still unsure if this path is the good one since wear leveling still probably affects also the LUKS header.  LUKS wiping is a quick option, but i'm not sure ti offers the level of security desired.

#436 would probably [do a better job](https://ata.wiki.kernel.org/index.php/ATA_Secure_Erase). Might want to wrap a script around different options.

PowerPress, 2019-05-01:
Wipe the LUKS header and then initiate Secure Erase than just a Secure Erase since a power cut could happen or any number of things before the erase gets to the sensitive data. Gives a belt and suspenders approach. Maybe during heads configuration provide an option for Just LUKS header or LUKS header and Secure Erase since not all drives support it though most modern ones will.

tlaurion, 2019-05-22:
PR welcome!

tlaurion, 2019-06-03:
@1989gironimo @PowerPress @jfrederickson

Commits of interest:
https://gitlab.com/kalilinux/packages/cryptsetup/commit/3ddbbb67ade4e60f4bcacd26d8647ae04c8fabb5
https://gitlab.com/kalilinux/packages/cryptsetup/commit/f4eb57ecde00ff55304ec8bc8684aab127198c35

tlaurion, 2019-06-26:
@PowerPress @1989gironimo @jfrederickson 

For the sake of clarifications:
[Disk Recovery Key is enforced in slot 0.](https://github.com/tlaurion/heads/blob/oem_reownership-staging_rebase_master_PR/initrd/bin/reencrypt-luks#L48)
[Disk Unlock Key is enforced in slot 1.](https://github.com/tlaurion/heads/blob/oem_reownership-staging_rebase_master_PR/initrd/bin/kexec-seal-key#L82)

Cryptsetup module has been upgraded to 1.7.5 [here](https://github.com/tlaurion/heads/commit/379823df1ec2dfbda153710c1a402bfac9ed634c#diff-46e6111a1a33bcff5c018e119dc2b14c)

[Cryptsetup patchset has been bonified](https://github.com/tlaurion/heads/blob/oem_reownership-staging_rebase_master_PR/patches/cryptsetup-1.7.5.patch) with wipe keyslots from [kali project](https://gitlab.com/kalilinux/packages/cryptsetup/blob/3ddbbb67ade4e60f4bcacd26d8647ae04c8fabb5/debian/patches/cryptsetup_nuke_keys.patch)

So implementing UX is missing. A logical implementation would be to implement [documented process ](https://www.kali.org/tutorials/nuke-kali-linux-luks/) in Heads Whiptail menus:

1. Offer an option to store the Wipe Slot Key in slot 2.
1. Offer an option to store LUKS header on USB
1. Offer an option to restore LUKS header from USB

tlaurion, 2020-03-09:
Patches needs to be ported to cryptsetup 1.7.5

tlaurion, 2020-07-22:
Note the Kali do not support the patchset anymore and changed their approach since the beginning of this discussion.
Documentation lies [here](https://www.kali.org/tutorials/nuke-kali-linux-luks/)

Basically, they decided to not patch cryptesetup and implement an external program called "cryptsetup-nuke-password" that has hooks in init. Please read full post referred above.

There is now a planning to upgrade cryptsetup under Heads to 2.x to support newer crypsetup made containers in newer OSes installation under #749.

I personally dropped to support this feature in my roadmap. Any help welcome.

closed 2020-08-01 by tlaurion