linuxboot/heads #445: Building coreboot for AGESA board fails

paulmenzel, 2018-09-05:
(Note, the problem happens also with coreboot 4.7 in the Heads master branch. Upgrading to coreboot 4.8.1 didn’t solve it.)

Trying to add the ASRock E350M1 to Heads, the coreboot build fails with the error below (from `build/log/coreboot.log`).

```
make -C "/mnt/build/coreboot-4.8.1/asrock-e350m1" -C /mnt/build/coreboot-4.8.1 obj=/mnt/build/coreboot-4.8.1/asrock-e350m1 DOTCONFIG=../../config/coreboot-asrock-e350m1.config BUILD_TIMELESS=1 CFLAGS_x86_32=-fdebug-prefix-map=/mnt=heads -gno-record-gcc-switches CFLAGS_x86_64=-fdebug-prefix-map=/mnt=heads -gno-record-gcc-switches
[…]
    LINK       cbfs/fallback/bootblock.debug
    OBJCOPY    cbfs/fallback/bootblock.elf
    OBJCOPY    bootblock.raw.bin
    CC         libagesa/vendorcode/amd/agesa/common/amdlib.o
    CC         libagesa/vendorcode/amd/agesa/f14/Legacy/Proc/Dispatcher.o
    CC         libagesa/vendorcode/amd/agesa/f14/Legacy/Proc/agesaCallouts.o
    CC         libagesa/vendorcode/amd/agesa/f14/Legacy/Proc/hobTransfer.o
    CC         libagesa/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.o
src/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.c: In function 'F14IsC6Supported':
src/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.c:113:3: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
   if ((((CPU_STATE_PM_CTRL1_REGISTER *) &PciRegister)->CoreC6Cap == 0) &&
   ^~
src/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.c:114:7: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
       (((CPU_STATE_PM_CTRL1_REGISTER *) &PciRegister)->PkgC6Cap == 0)) {
       ^
src/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.c: In function 'F14InitializeC6':
src/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.c:152:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     if (((PSTATE_MSR *) &MsrReg)->PsEnable == 1) {
     ^~
src/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.c:163:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     ((CPU_STATE_PM_CTRL1_REGISTER *) &PciRegister)->CoreC6Dis = 0;
     ^
src/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.c:164:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     ((CPU_STATE_PM_CTRL1_REGISTER *) &PciRegister)->PkgC6Dis = 0;
     ^
src/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.c:173:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     ASSERT (((MEM_CFG_LOW_REGISTER *) &PciRegister)->C6DramLock == 1);
     ^~~~~~
src/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.c:177:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     ASSERT (((C6_BASE_REGISTER *) &PciRegister)->C6Base != 0);
     ^~~~~~
src/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.c:184:5: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
     if (((CPU_STATE_PM_CTRL1_REGISTER *) &PciRegister)->PkgC6Cap == 1) {
     ^~
src/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.c:185:7: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
       ((CPU_STATE_PM_CTRL1_REGISTER *) &PciRegister)->PstateIdCoreOffExit = MaxEnabledPstate;
       ^
cc1: all warnings being treated as errors
make[1]: *** [Makefile:338: asrock-e350m1/libagesa/vendorcode/amd/agesa/f14/Proc/CPU/Family/0x14/F14C6State.o] Error 1
make[1]: Leaving directory '/mnt/build/coreboot-4.8.1'
```

The problem seems to be, that the CFLAGS for AGESA or somehow ignored.

The main `Makefile.inc` enables the warning.

    Makefile.inc:CFLAGS_common += -Wstrict-aliasing -Wshadow -Wdate-time

But in `src/vendorcode/amd/agesa/f14/Makefile.inc` there is:

    CFLAGS_x86_32 += -march=k8-sse3 -mtune=k8-sse3 -fno-zero-initialized-in-bss -fno-strict-aliasing
    CFLAGS_x86_64 += -march=k8-sse3 -mtune=k8-sse3 -fno-zero-initialized-in-bss -fno-strict-aliasing

Building coreboot manually, works fine.

paulmenzel, 2018-09-05:
@jorgefm, as the AMD Bettong board also uses AGESA, I wonder if you hit the same problem.

[1]: https://github.com/osresearch/heads/issues/440

paulmenzel, 2018-09-05:
Heads’ coreboot module seems to override the CFLAGS in an incompatible way.

```
coreboot_target := \
        -C $(build)/$(coreboot_base_dir) \
        obj=$(build)/$(coreboot_dir) \
        DOTCONFIG=../../$(CONFIG_COREBOOT_CONFIG) \
        BUILD_TIMELESS=1 \
        CFLAGS_x86_32="$(EXTRA_FLAGS)" \
        CFLAGS_x86_64="$(EXTRA_FLAGS)" \
        $(MAKE_JOBS)
```

jorgefm, 2018-09-05:
Hi,
I've not hit this problem because the Bettong demoboard is compiling against other AGESA family, exactly this one:

    CC         libagesa/vendorcode/amd/pi/00660F01/Proc/Fch/Common/FchLib.o
    CC         libagesa/vendorcode/amd/pi/00660F01/Proc/Fch/Common/FchPeLib.o
    CC         libagesa/vendorcode/amd/pi/00660F01/Proc/Fch/Common/MemLib.o
    CC         libagesa/vendorcode/amd/pi/00660F01/Proc/Fch/Common/PciLib.o
    CC         libagesa/vendorcode/amd/pi/00660F01/Proc/Psp/PspBaseLib/PspBaseLib.o
    CC         libagesa/vendorcode/amd/pi/00660F01/binaryPI/AGESA.o
    CC         libagesa/vendorcode/amd/pi/Lib/amdlib.o
    CC         libagesa/vendorcode/amd/pi/Lib/imc/HwmLateService.o
    CC         libagesa/vendorcode/amd/pi/Lib/imc/ImcLib.o
    AGESA      libagesa/libagesa.a

paulmenzel, 2018-09-07:
Not setting CFLAGS to `EXTRA_FLAGS` fixed the build issue.

```
diff --git a/modules/coreboot b/modules/coreboot
index 99637f6..b60cd2e 100644
--- a/modules/coreboot
+++ b/modules/coreboot
@@ -16,24 +16,18 @@ CONFIG_COREBOOT_CONFIG ?= config/coreboot-$(BOARD).config
 # Ensure that touching the config file will force a rebuild
 $(build)/$(coreboot_dir)/.configured: $(CONFIG_COREBOOT_CONFIG)
 
-EXTRA_FLAGS := -fdebug-prefix-map=$(pwd)=heads -gno-record-gcc-switches
-
 coreboot_configure := \
        $(MAKE) -C $(build)/$(coreboot_base_dir) \
                oldconfig \
                obj=$(build)/$(coreboot_dir) \
                DOTCONFIG=../../$(CONFIG_COREBOOT_CONFIG) \
-               BUILD_TIMELESS=1 \
-               CFLAGS_x86_32="$(EXTRA_FLAGS)" \
-               CFLAGS_x86_64="$(EXTRA_FLAGS)" \
+               BUILD_TIMELESS=1
 
 coreboot_target := \
        -C $(build)/$(coreboot_base_dir) \
        obj=$(build)/$(coreboot_dir) \
        DOTCONFIG=../../$(CONFIG_COREBOOT_CONFIG) \
        BUILD_TIMELESS=1 \
-       CFLAGS_x86_32="$(EXTRA_FLAGS)" \
-       CFLAGS_x86_64="$(EXTRA_FLAGS)" \
        $(MAKE_JOBS)
 
 coreboot_output := coreboot.rom
```

No, idea what the right solution is.