linuxboot/heads #447: kexec-select-boot: Find a way to not search whole partition

paulmenzel, 2018-09-08:
With devices, where `/boot` is not on a separate partition, but also on the / partition, the find call takes a very long time.

      for i in `find $bootdir -name "$config"`; do
              kexec-parse-boot "$bootdir" "$i" >> $option_file
      done

Passing `-b /boot/boot` does not work, as the Linux image is only searched in `/boot`.

No idea, what an elegant solution would be. Maybe the path to `/boot/grub` (`/boot/boot/grub`) should be made configurable?

rminnich, 2018-09-08:
how long is long? What is the media? is this an intrinsic issue with traversing the file system or is find slow?

paulmenzel, 2018-09-08:
Am 08.09.2018 um 18:09 schrieb ron minnich:
> how long is long? What is the media?

It’s a 64 GB AHCI SSD connected over SATA. I just rebooted after waiting
ten seconds. I did not try the 1 TB HDD yet.

> is this an intrinsic issue with traversing the file system or is find
> slow?

I’d say it’s wrong to search anything else then the files, under `/boot` 
(in the end file system, not initramfs).

flammit, 2018-09-08:
Thanks @paulmenzel! The searching was set to be quite permissive because of the wide variation of boot config files on install/live distro ISOs. Agree though that on the fixed media paths we can be much more selective and assume grub/freedesktop style paths

merge, 2018-09-25:
while finding boot options is slow in this case, it's more extreme when it comes to hashing. hashing the whole system is not desired in this case.

Thrilleratplay, 2020-09-28:
`/boot` is assumed to be the boot partition. It is not the most elegant but what about where ever `kexec-select-boot` passes in `/boot` replace it with a variable and check if there 

```sh
bootdir= "/boot"
if [ -d "/boot/boot/" ]; then
  bootdir= "/boot/boot"
fi
kexec-select-boot -m -b "$bootdir" -c "grub.cfg"
```

There a few places where checking for the specific boot directory would be useful like the oem factory reset where all files in are run through `sha256sum`

https://github.com/osresearch/heads/blob/master/initrd/bin/oem-factory-reset#L122