linuxboot/heads #450: Introduce option for specifying path for Option ROMs

paulmenzel, 2018-09-08:
The Lenovo ThinkPad X230 can use a VGA Option ROM (Video BIOS). AMD’s integrated graphics devices systems actually require these (ATOMBIOS).

Currently, the file needs to be in the coreboot build directory, if a relative path is specified.

An elegant solution should be found, how this can be better integrated into the board configuration file, and is put into coreboot from there.

tlaurion, 2018-09-09:
What is the advantage of using option vga rom when there is native vga init
for the x230?

paulmenzel, 2018-09-10:
Am 10.09.2018 um 00:49 schrieb tlaurion:
> What is the advantage of using option vga rom when there is native vga init
> for the x230?

The default Lenovo ThinkPad X230 coreboot configuration file, seems to 
list the VGA Option ROM.

Anyway, I wrote *can* for the Lenovo ThinkPad X230, and *require* for 
AMD systems. The issue is mainly for the AMD graphics devices.

tlaurion, 2024-04-16:
Dgpu boards can be seen as reference for other board inclusion.

closed 2024-04-16 by tlaurion