linuxboot/heads #451: x230-flash not building (too big)

mfc, 2018-09-09:
i am not able to make `BOARD=x230-flash`, this is the error i get while trying make `x230-flash` on debian-9 (using newest heads code commit ` 1d27c93`). if the issue is the make version (`Wrong make detected: 4.1`) let me know how to force it to use 4.2. thanks!

```
make BOARD=x230-flash
Wrong make detected: 4.1
LANG=C MAKE=/home/user/heads/build/make-4.2/make /home/user/heads/build/make-4.2/make  all
make[1]: Entering directory '/home/user/heads'
2018-09-10 00:24:45+02:00 INSTALL-MODULE drivers/net/ethernet/intel/e1000e/e1000e.ko
2018-09-10 00:24:45+02:00 INSTALL-MODULE drivers/usb/host/ehci-hcd.ko
2018-09-10 00:24:45+02:00 INSTALL-MODULE drivers/usb/host/ehci-pci.ko
2018-09-10 00:24:45+02:00 INSTALL-MODULE drivers/usb/host/xhci-hcd.ko
2018-09-10 00:24:45+02:00 INSTALL-MODULE drivers/usb/host/xhci-pci.ko
2018-09-10 00:24:45+02:00 INSTALL-MODULE drivers/usb/storage/usb-storage.ko
2018-09-10 00:24:45+02:00 SYMLINK bin/busybox
2018-09-10 00:24:50+02:00 INSTALL-BIN build/pciutils-3.5.4/lspci
2018-09-10 00:24:50+02:00 INSTALL-BIN build/flashrom-1.0/flashrom
2018-09-10 00:24:50+02:00 INSTALL-BIN build/gnupg-1.4.21/g10/gpg
2018-09-10 00:24:50+02:00 INSTALL-BIN build/dropbear-2016.74/ssh
2018-09-10 00:24:50+02:00 INSTALL-BIN build/dropbear-2016.74/scp
2018-09-10 00:24:50+02:00 INSTALL-BIN build/dropbear-2016.74/dropbear
2018-09-10 00:24:50+02:00 INSTALL-BIN build/flashtools-0.0.1/flashtool
2018-09-10 00:24:50+02:00 INSTALL-BIN build/flashtools-0.0.1/peek
2018-09-10 00:24:50+02:00 INSTALL-BIN build/flashtools-0.0.1/poke
2018-09-10 00:24:50+02:00 INSTALL-BIN build/flashtools-0.0.1/cbfs
2018-09-10 00:24:50+02:00 INSTALL-BIN build/flashtools-0.0.1/uefi
2018-09-10 00:24:50+02:00 INSTALL-BIN build/coreboot-4.7/util/cbmem/cbmem
2018-09-10 00:24:50+02:00 INSTALL-LIB build/pciutils-3.5.4/lib/libpci.so.3.5.4
2018-09-10 00:24:50+02:00 INSTALL-LIB build/pciutils-3.5.4/../../install/lib/libpci.so.3
2018-09-10 00:24:50+02:00 INSTALL-LIB build/libusb-1.0.21/libusb/.libs/libusb-1.0.so.0
2018-09-10 00:24:50+02:00 INSTALL-LIB build/musl-1.1.15/lib/libc.so
2018-09-10 00:24:50+02:00 INSTALL-LIB build/libusb-compat-0.1.5/libusb/.libs/libusb-0.1.so.4
2018-09-10 00:24:50+02:00 INSTALL-LIB build/zlib-1.2.11/libz.so.1
2018-09-10 00:24:50+02:00 INSTALL boards/x230-flash/x230-flash.config
2018-09-10 00:24:50+02:00 HASH 1d2dfd4affdc773329550e694c274cff8e71639b clean x230-flash
2018-09-10 00:24:50+02:00 CPIO      build/x230-flash/tools.cpio
2018-09-10 00:24:50+02:00 UNCHANGED build/x230-flash/tools.cpio
2018-09-10 00:24:50+02:00 CPIO      build/x230-flash/heads.cpio
2018-09-10 00:24:50+02:00 UNCHANGED build/x230-flash/heads.cpio
2018-09-10 00:24:50+02:00 MAKE coreboot
tail /home/user/heads/build/log/coreboot.log
-----
make[2]: Entering directory '/home/user/heads/build/coreboot-4.7'
    GEN        generated/bootblock.ld
    CP         bootblock/arch/x86/bootblock.ld
    LINK       cbfs/fallback/bootblock.debug
    OBJCOPY    cbfs/fallback/bootblock.elf
    OBJCOPY    bootblock.raw.bin
Created CBFS (capacity = 4194008 bytes)
    CBFS       fallback/romstage
    CBFS       mrc.cache
    CBFS       cpu_microcode_blob.bin
    CBFS       fallback/ramstage
    CBFS       cmos_layout.bin
    CBFS       fallback/dsdt.aml
    CBFS       fallback/payload
E: Could not add [../../build/x230-flash/bzImage, 3914701 bytes (3822 KB)@0x0]; too big?
E: Failed to add '../../build/x230-flash/bzImage' into ROM image.
E: Failed while operating on 'COREBOOT' region!
E: The image will be left unmodified.
make[2]: *** [Makefile.inc:848: x230-flash/coreboot.pre] Error 1
make[2]: Leaving directory '/home/user/heads/build/coreboot-4.7'
make[1]: *** [Makefile:356: /home/user/heads/build/coreboot-4.7/x230-flash/.build] Error 1
make[1]: Leaving directory '/home/user/heads'
Makefile:559: recipe for target 'all' failed
```

mfc, 2018-09-10:
will that prevent me from configuring U2F/Yubikey?

tlaurion, 2018-09-10:
@mfc x230-flash use is only to permit internal flash of the x230.rom.

There is no current use of GPG being integrated into x230-flash.

tlaurion, 2018-09-10:
I would suggest to use that branch for the moment, so that you benefit from fbwhiptail and core boot 4.8.1 and other merges that were not integrated from master.

https://github.com/tlaurion/heads/tree/x230-current-No_Librem_key-Working-coreboot-4.8

flawedworld, 2018-09-14:
@mfc Try increase cbfs size

merge, 2018-10-11:
While it *could be* nice to have a 4M coreboot build with flashrom on a shell included, IMO x230-flash is a mess and causes too much confusion :)

I'd do away with it completely and document first-time installation more "generically" using runtime-Linux, or not at all. Upgrading, as in http://osresearch.net/Upgrading could probably deserve to become more streamlined though.

tlaurion, 2018-10-12:
@merge : seriously, I think x230-flash should still exist but be minimal and used only to call something like [flash-gui.sh](https://github.com/osresearch/heads/blob/master/initrd/bin/flash-gui.sh) to propose which firmware to flash 8MB and nothing else.

The result of this would be easy to use by anybody UX wise.

That implies a cleaning of dependencies of [FBWHIPTAIL](https://github.com/osresearch/heads/blob/master/modules/fbwhiptail) since the image is still too big to fit, with weird dependencies not needed for this use case. 

TODO: Will report back. Tried to reduce x230-flash footprint before but was facing issues deactivating more modules under [x230-flash](https://github.com/tlaurion/heads/blob/current_x230_no-CONFIG_LIBREMKEY-working_GPG2/boards/x230-flash/x230-flash.config) while [introducing FBWHIPTAIL dependencies](https://github.com/tlaurion/heads/blob/current_x230_no-CONFIG_LIBREMKEY-working_GPG2/boards/x230/x230.config#L20) back into x230-flash board config)

tlaurion, 2018-10-12:
@flawedworld : 
> Try increase cbfs size

Tried in the past, but now it would be handy:
```
user@build-x230:~/Downloads/heads$ make BOARD=x230-flash -j4
2018-10-12 16:02:47-04:00 INSTALL-MODULE drivers/usb/host/ehci-pci.ko
2018-10-12 16:02:47-04:00 INSTALL-MODULE drivers/usb/host/ehci-hcd.ko
2018-10-12 16:02:47-04:00 INSTALL-MODULE drivers/usb/host/xhci-pci.ko
2018-10-12 16:02:47-04:00 INSTALL-MODULE drivers/usb/host/xhci-hcd.ko
2018-10-12 16:02:47-04:00 SYMLINK bin/busybox
2018-10-12 16:02:47-04:00 INSTALL-MODULE drivers/usb/storage/usb-storage.ko
2018-10-12 16:02:47-04:00 INSTALL-BIN build/pciutils-3.5.4/lspci
2018-10-12 16:02:47-04:00 INSTALL-BIN build/flashrom-1.0/flashrom
2018-10-12 16:02:47-04:00 INSTALL-BIN build/fbwhiptail/fbwhiptail
2018-10-12 16:02:47-04:00 INSTALL-BIN build/fbwhiptail/whiptail
2018-10-12 16:02:47-04:00 MAKE cbmem
make[1]: Entering directory '/home/user/Downloads/heads/build/coreboot-4.8.1/util/cbmem'
2018-10-12 16:02:47-04:00 INSTALL-LIB build/libpng-1.6.34/.libs/libpng16.so.16
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/user/Downloads/heads/build/coreboot-4.8.1/util/cbmem'
2018-10-12 16:02:47-04:00 INSTALL-LIB build/cairo-1.14.12/src/.libs/libcairo.so.2
2018-10-12 16:02:47-04:00 INSTALL-LIB build/pixman-0.34.0/pixman/.libs/libpixman-1.so.0
2018-10-12 16:02:47-04:00 INSTALL-LIB build/pciutils-3.5.4/lib/libpci.so.3.5.4
2018-10-12 16:02:47-04:00 INSTALL-LIB build/musl-1.1.15/lib/libc.so
2018-10-12 16:02:47-04:00 INSTALL-LIB build/pciutils-3.5.4/../../install/lib/libpci.so.3
2018-10-12 16:02:47-04:00 CPIO      build/x230-flash/heads.cpio
2018-10-12 16:02:47-04:00 INSTALL-LIB build/zlib-1.2.11/libz.so.1
2018-10-12 16:02:47-04:00 INSTALL boards/x230-flash/x230-flash.config
2018-10-12 16:02:47-04:00 INSTALL-BIN build/coreboot-4.8.1/util/cbmem/cbmem
2018-10-12 16:02:47-04:00 HASH 20615be9af9de637db09bac79dd9fb32cf223d6d dirty x230-flash
2018-10-12 16:02:47-04:00 UNCHANGED build/x230-flash/heads.cpio
2018-10-12 16:02:52-04:00 CPIO      build/x230-flash/tools.cpio
2018-10-12 16:02:52-04:00 UNCHANGED build/x230-flash/tools.cpio
2018-10-12 16:02:52-04:00 MAKE coreboot
tail /home/user/Downloads/heads/build/log/coreboot.log
-----
make[1]: Entering directory '/home/user/Downloads/heads/build/coreboot-4.8.1'
error: pathspec '3rdparty/blobs' did not match any file(s) known to git.
    GEN        generated/bootblock.ld
    CP         bootblock/arch/x86/bootblock.ld
    LINK       cbfs/fallback/bootblock.debug
    OBJCOPY    cbfs/fallback/bootblock.elf
    OBJCOPY    bootblock.raw.bin
Created CBFS (capacity = 4063192 bytes)
    CBFS       fallback/romstage
    CBFS       cpu_microcode_blob.bin
    CBFS       fallback/ramstage
    CBFS       cmos_layout.bin
    CBFS       fallback/dsdt.aml
    CBFS       fallback/payload
E: Could not add [../../build/x230-flash/bzImage, 4191645 bytes (4093 KB)@0x0]; too big?
E: Failed to add '../../build/x230-flash/bzImage' into ROM image.
E: Failed while operating on 'COREBOOT' region!
E: The image will be left unmodified.
make[1]: *** [Makefile.inc:920: x230-flash/coreboot.pre] Error 1
make[1]: Leaving directory '/home/user/Downloads/heads/build/coreboot-4.8.1'
make: *** [Makefile:356: /home/user/Downloads/heads/build/coreboot-4.8.1/x230-flash/.build] Error 1
```

merge, 2018-10-15:
increasing the `x230-flash` cbfs size is not an option. The whole point of x230-flash is to be 4M. That's how we enable easy hardware-flashing / first-time bootstrapping.

for `x230` it's 7M and could be increased even further if we choose to depend on me_cleaner being applied (which would add some more complexity to first-time flashing of course).

tlaurion, 2018-10-15:
@merge : any idea how to get that little additional space needed to have CAIRO, FBWHIPTAIL in to directly call flash-gui.sh from init from x230-flash 4MB rom?

merge, 2018-10-16:
at first sight from your logs, the microcode update comes to mind...

tlaurion, 2018-11-08:
Or maybe whiptail only without FBWhiptail. Will try!

tlaurion, 2018-11-11:
@nemanjan00 : simply comment out CONFIG_GPG under board/x230-flash/x230-flash.conf

nemanjan00, 2018-11-11:
Thanks, integrated that into my CI. :D

https://gitlab.com/nemanjan00/heads-ci/-/jobs/119040631

nemanjan00, 2018-11-13:
Fixed