linuxboot/heads #456: circleci status on main page

osresearch, 2018-09-18:
We should display the circleci status and last hashes on the mainpage

rjoleary, 2018-09-19:
You can put this into the README to show the build status:

[![Build
Status](https://circleci.com/gh/osresearch/heads/tree/master.png?style=shield)](https://circleci.com/gh/osresearch/heads/tree/master)

```
[![Build
Status](https://circleci.com/gh/osresearch/heads/tree/master.png?style=shield)](https://circleci.com/gh/osresearch/heads/tree/master)
```

tlaurion, 2019-01-27:
@osresearch : That would be awesome. With artifact downloads :)

Is there other deployed CI? I know of the following ones:
https://gitlab.com/nemanjan00/heads-ci/-/jobs
https://circleci.com/gh/osresearch/heads

closed 2020-08-01 by tlaurion