linuxboot/heads #463: Deploy cryptsetup-reencrypt in Heads to permit QubesOS image deployment in organization

tlaurion, 2018-10-06:
Discussion [here](https://github.com/QubesOS/qubes-issues/issues/1906#issuecomment-422142622) for technical specifics and [here](https://github.com/SkypLabs/my-qubes-os-formula/issues/6#issue-364582625) for rationale.

tlaurion, 2018-10-06:
To make cryptsetup-reencrypt available in heads:
```
diff --git a/modules/cryptsetup b/modules/cryptsetup
index e81d356..4cea7f3 100644
--- a/modules/cryptsetup
+++ b/modules/cryptsetup
@@ -15,6 +15,7 @@ cryptsetup_configure := ./configure \
 	--host i386-elf-linux \
 	--prefix "/" \
 	--disable-gcrypt-pbkdf2 \
+	--enable-cryptsetup-reencrypt \
 	--with-crypto_backend=kernel \
 
 # but after building, replace prefix so that they will be installed
@@ -28,6 +29,7 @@ cryptsetup_target := \
 
 cryptsetup_output := \
 	src/.libs/cryptsetup \
+	src/.libs/cryptsetup-reencrypt \
 	src/.libs/veritysetup \
 
 cryptsetup_libraries := \

```

tlaurion, 2018-10-06:
Reencrypting after wiping out slot 1 used by TPMTOTP released key.
Estimated reencrytion time is 30 minutes on Intel SSD without any optimized settings, average speed is 80MiB/s.

tlaurion, 2018-10-06:
Slows down to average 54MiB/s after a while and cruises there. Might need to tweak reencrypt settings with something like `[cryptsetup-reencrypt -B 32 --use-directio /dev/sda2](https://asalor.blogspot.com/2012/08/re-encryption-of-luks-device-cryptsetup.html)`.

tlaurion, 2018-10-08:
Result report:

`cryptsetup-reencrypt /dev/sda2`: reencryption took 56m to reencrypt 151600MiB at an everage speed of 45.4 MiB/s
`cryptsetup-reencrypt -B 32 --use-directio /dev/sda2` : reencryption took 25m to reencrypt 151600 MiB at an average speed of 98.3 MiB/s

tlaurion, 2018-10-17:
[From @kylerankin](https://github.com/osresearch/heads/pull/464#issuecomment-429913562): 
> I'd love to have an OEM-ready Qubes installer. The challenge with Qubes OEM images in the past was having resources with the expertise to build it. The Qubes team could do it but you would need to be willing to set up a consulting agreement with them and fund the effort.

osresearch, 2018-11-07:
Was this resolved by merging #464 or is there more to be done?

tlaurion, 2018-11-08:
https://github.com/osresearch/heads/issues/475 a whiptail menu permitting to reencrypt drive is still missing.

tlaurion, 2019-01-27:
Included in #511

tlaurion, 2019-05-01:
fixed in #551

closed 2019-05-01 by tlaurion