linuxboot/heads #473: Behavior not as expected after flashing 12MB Heads ROM

jfrederickson, 2018-11-06:
Hey, so I'm a bit lost when trying to install Heads here... I've flashed the modified 12MB Heads ROM with my public key included (through `flashrom-x230.sh`, and taken ownership of the TPM. The installation guide seems to suggest that I'd get a boot menu at this point, but that doesn't appear to be the case; this is all I get when I boot with a Qubes flash drive inserted.

[image]

I'm not entirely sure how to proceed at this point as I'm not super familiar with how Heads is supposed to work. That said I'm also not confident that I've installed my GPG keys correctly; I couldn't find the `cbfs` tool on my Heads installation, so I used `cbfstool` from my current laptop (which doesn't appear to take the same arguments as described in the installation guide).

flammit, 2018-11-06:
howdy @jfrederickson!  the current docs are written for the master branch as the latest binary release is quite old and lacking in many features.  You should be able to follow http://osresearch.net/Building (essentially it's just a `make BOARD=x230` after cloning).  Let me know if you run into any issues or ask for help in the u-root slack #heads channel (http://slack.u-root.com/ for the invite)

jfrederickson, 2018-11-06:
Ahh, that would explain it! Thanks, I'll give that a try! :)

flammit, 2018-11-06:
Yup, you should be able to use the old `flashrom-x230.sh` with the new 12MB rom file to update.  As always though, it's best to have the external flash option to recover from bricks.  I've actually always just flashed from beagleboneblack/rpi instead of the WOL version, but maybe I've just not tested my luck enough.

jfrederickson, 2018-11-06:
Yeah, I tried for a while to get an external flasher running, but without any success. That's how I flashed my x200 with Libreboot, but... eh, I figure I'll do things this way until I screw something up and have to get it working. I won't be able to use the WoL method again as I cleaned ME the first time, so I've got one shot. :P

merge, 2019-05-22:
Seems like this issue can safely be closed.

closed 2019-06-22 by tlaurion