linuxboot/heads #477: Generalize whiptail usage to all platforms

tlaurion, 2018-11-09:
I think it might be a good time to replace generic-init by gui-init, and let the board configuration options define if FBWhiptail is desired. I think that it would be a good base in improving the UX for everyone.

@kylerankin @osresearch @flammit : What do you think?

tlaurion, 2019-02-08:
@merge : 
Yes. But each board should be validated individually to have proper Linux config FB support first from people having the models.

For the x230 board to become compatible with FBWhiptail, here were the [required changes](https://github.com/osresearch/heads/commit/98598e7f2f27eeb831a66eeb3f50d598c9ff50d0).
To have Whiptail (console based) instead of FBwhiptail (FB based):

```
#Whiptail
CONFIG_SLANG=y
CONFIG_NEWT=y
#FBWhiptail
#CONFIG_CAIRO=y
#CONFIG_FBWHIPTAIL=y

#BOOT SCRIPT SELECTION
#export CONFIG_BOOTSCRIPT=/bin/generic-init
export CONFIG_BOOTSCRIPT=/bin/gui-init
#export CONFIG_BOOTSCRIPT_NETWORK=/bin/network-init-recovery
```
This is the list of board owners i've gathered from board config changes "blames". Can you test the above changes on the board config on those boards and adapt linux config files to activate FB and report?

@osresearch : leopard, tioga
@vejmarie : r630, s2600wf
@jgrip @flammit @osresearch : x220

osresearch, 2019-02-08:
leopard/tioga have no video cards at all, r630/s2600wf are almost always headless.

tlaurion, 2019-02-08:
So something like for the kgpe-d16 for all those.

closed 2025-09-22 by tlaurion