linuxboot/heads #485: X230 Build does not include latest microcode from Intel, possible security issue.

JeffreyAbbinante, 2018-11-23:
Title says pretty much all. Building heads on an X230 will not include the latest version of the intel microcode. The microcode Heads ships with is 0x1f (31) but the latest version is 0x20 (32). This is an issue because version 0x20 patches out even more spectre related vulnerabilities. The problem is the coreboot upstream does not support the latest version of the microcode yet. Is there a way to support it in heads similar to how the people behind skulls have?

pgeorgi, 2018-11-24:
Adding that update is discussed on https://review.coreboot.org/c/blobs/+/27352

merge, 2018-12-13:
This is not a Heads issue as we don't use Linux to load microcode updates and usually don't use out-of-tree coreboot changes, like we allow it in the Skulls build (that does nothing really itself). Heads has enough other problems to solve in order to achieve it's goal someday :) In a sense we can't deal with other project's details here :)

@pgeorgi Even for coreboot everything is on track: your mentioned change is deprecated and said update is already merged in the blobs repository by https://review.coreboot.org/c/blobs/+/30067 so only the coreboot repository needs to bump up the submodule pointer, which will afterwards result in Heads having version 0x20.

Looks like this issue can safely be closed.

JeffreyAbbinante, 2018-12-20:
> This is not a Heads issue as we don't use Linux to load microcode updates and usually don't use out-of-tree coreboot changes, like we allow it in the Skulls build (that does nothing really itself). [...]

I agree that this can be closed, when the upstream bumps up the microcode to 0x20. I was wondering why I couldn't just "stitch in" the microcode into the Heads ROM via cbfstool (I bricked my laptop trying to put Skulls microcode into the Heads firmware). Thanks for the insight as to how Heads handles microcode updates. The concern is for users of operating systems that do not include microcode updates such as PureOS or Debian. For now, would a user interested in keeping microcode updates away from the OS want to either stick with Skulls or is there another way to update Heads microcode by hand?

merge, 2018-12-21:
@JeffreyAbbinante export/import using cbfstool seems dangerous. _If_ at all, you should patch coreboot during Building of Heads.

Also, Debian provides microcode updates in non-free, and lets Linux load them automatically during boot.

Also, the upstream already _did_ bump up to rev. 0x20. It only a matter of release-procedures, tagging and time. No action required by Heads here.

Finally, I hope you can recover you laptop :)

JeffreyAbbinante, 2018-12-21:
Thanks so much for the explanation, Martin @merge . See it was 3 am one night and I was being an idiot. I was able to get Libreboot X200 to accept microcode updates via the cbfstool method (I was able to get a newer version that what Debian offered in non-free). Luckily, the only issue was that I was facing kernel panics due to the lack of microcode. I had to drive back to my university to get my SPI programmer (rpi 3 with a pomona clip) and loaded your Skulls ROM on there. It seems that the solution is to wait for an update.

JeffreyAbbinante, 2018-12-21:
Issue has been solved upstream. Waiting for project to merge changes.

closed 2018-12-21 by JeffreyAbbinante