linuxboot/heads #498: Pin musl-cross-git to a particular commit for reproducible builds

strugee, 2018-12-14:
While patching musl-cross it came to my attention that Heads' build script just clones the latest git commit. This seems like a problem for reproducible builds? Especially since upstream can break our build. In fact I first noticed this because I pushed a commit to my fork (which I patched Heads to pull from) and my Heads build started failing because patching the musl-cross config file failed.

strugee, 2018-12-14:
Note that I just pulled the latest commit from GitHub for all of these, which I assume is the right thing to do?

tlaurion, 2019-03-20:
Discussion follows in [pull request](https://github.com/osresearch/heads/pull/499#issuecomment-474998516)

tlaurion, 2019-04-13:
@strugee : any base Makefile change proposition to pull the code upon detected git changes instead? That would do it IMHO.

tlaurion, 2019-05-21:
#567 seems to be the solution for this.
@strugee ?

closed 2020-05-24 by tlaurion