linuxboot/heads #500: Update coreboot from 4.8.1 to 4.9

tlaurion, 2018-12-21:


JeffreyAbbinante, 2018-12-21:
Refer to pull request #501

tlaurion, 2018-12-21:
@JeffreyAbbinante : 
```
cd heads
cp -r patches/coreboot-4.8.1/ patches/coreboot-4.9 
```
Start the work :)

JeffreyAbbinante, 2018-12-21:
What patches should I be leaving out? I only have an X230 to test them on.

tlaurion, 2018-12-21:
@JeffreyAbbinante : 0000 - measured boot needs to be figured out for sure.

JeffreyAbbinante, 2018-12-21:
Can you contribute to my pull request? I can add you to my fork if you like. For now, I have coreboot 4.9 booting with the heads system.

flammit, 2018-12-22:
@JeffreyAbbinante FYI what @tlaurion is referring to is that the `patches/` directory is version-specific (`patches/coreboot-4.8.1`), so while you've updated the coreboot module to be 4.9, the patches to coreboot that provide Heads-specific functionality need to be migrated and tested as well.  Right now, your system is booting using vanilla coreboot without the module measurement, etc.

JeffreyAbbinante, 2018-12-22:
@flammit I understood what he meant.  I am compiling a version with the patches already right now.

flammit, 2018-12-22:
@JeffreyAbbinante whoops - that's what I get for not reading the whole thread.  sorry about that and thanks for the help!

JeffreyAbbinante, 2018-12-22:
@flammit no problem. Would anyone happen to know what patches were merged into upstream?

tlaurion, 2018-12-22:
@JeffreyAbbinante :

I'm unfortunately doing something else right now and can't review the patches. I've started, but since I did not see measure boot functions as I was expecting and the measured patch failed, I went to d other things after asking @zaolin for more information.

Here is what I have in my bash history to test applicability of each patch:
```
cd heads
find ./ -name *_verify|xargs rm
tar cvzf packages-coreboot49.tar.gz  ./build/coreboot-4.9/util/crossgcc/tarballs/ packages/
cp -r patches/coreboot-4.8.1/ patches/coreboot-4.9
rm -rf build/coreboot-4.9/ && tar zxvf packages-coreboot49.tar.gz && make BOARD=x230
```

The patches will be applied one one by one when building the coreboot module you modified already.
Then one by one, verify what failed in .rej files produced, and modify original patches accordingly. I will resume this work in the next following days but i'm really happy someone else is also working on it :)

Let us know where it hangs!

JeffreyAbbinante, 2018-12-25:
So I took out my other laptop with a bit more processing power... to make compiling faster. I hope to find some stuff later this week.

JeffreyAbbinante, 2018-12-27:
The measured boot patch needs work for sure.
"Applying patch file : patches/coreboot-4.9/0000-measuredboot.patch 
patching file src/Kconfig
Hunk #1 succeeded at 278 (offset 18 lines).
can't find file to patch at input line 31
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git ./src/drivers/pc80/tpm/romstage.c ./src/drivers/pc80/tpm/romstage.c
|index b8e4705..7732e66 100644
|--- ./src/drivers/pc80/tpm/romstage.c
|+++ ./src/drivers/pc80/tpm/romstage.c
--------------------------
File to patch:

JeffreyAbbinante, 2018-12-27:
0052-crossgcc-Update-to-clang-6.0-cmake-3.11.3.patch has already been applied
Applying patch file : patches/coreboot-4.9/0052-crossgcc-Update-to-clang-6.0-cmake-3.11.3.patch 
patching file util/crossgcc/buildgcc
Reversed (or previously applied) patch detected!  Assume -R? [n]

tlaurion, 2018-12-29:
@zaolin confirms that measured boot hasn't made it to coreboot 4.9 as planned.

MrChromebox, 2019-02-03:
0007-intel-fsp-fsp2_0-Fix-FSP-2.0-headers-to-match-github.patch

should be dropped, since coreboot now has the option to use the headers/binaries from the 3rdparty/fsp repo (which is a mirror of the Intel FSP github repo). Just requires changing the coreboot config for affected boards, which is trivial.

tlaurion, 2019-02-08:
Let's track [measured boot WiP upstream in coreboot](https://review.coreboot.org/c/coreboot/+/29547/43) and switch to coreboot git once it's merged?

tlaurion, 2019-02-10:
refer to #515

tlaurion, 2019-03-23:
- [ ] integrate/ test : 
    - [ ] [Add VBOOT support](https://review.coreboot.org/c/coreboot/+/29816) 
    - [ ] [Enable VBOOT support](https://review.coreboot.org/c/coreboot/+/29815/1)
  - [ ] check if fmap is correct
  - [ ] set CMOS flags correctly

merge, 2019-03-25:
> To be added in this [PR](https://github.com/osresearch/heads/pull/515)
> 
>     * [ ]  integrate/ test :
>       
>       * [ ]  [Add VBOOT support](https://review.coreboot.org/c/coreboot/+/29816)
>       * [ ]  [Enable VBOOT support](https://review.coreboot.org/c/coreboot/+/29815/1)
> 
>     * [ ]  check if fmap is correct
> 
>     * [ ]  set CMOS flags correctly

does this apply to the X230 too? Also, how about being lazy and bump coreboot with the next release that includes measured boot in vboot +  support for the X230? I guess we'll have to do changes in userspace anyways? do we? which ones and who will do them? thanks!

tlaurion, 2019-03-26:
@merge: I have none of the answers right now. I just added information here to update status. From my understanding, those changes would work with thrbx230 as well but need to be tested.

Yep, userspace will need to be adjusted also, since only a subset of PCRs are used now. I will find relative information on u-root and edit this post later on.

@zaolin: your direct input would be more then welcome on the work needing to be done.

tlaurion, 2019-03-27:
@merge : from https://review.coreboot.org/c/coreboot/+/29547/73/Documentation/security/vboot/measured_boot.md

userland tools will have to take into account sealing of only the following PCRs:
```

## Platform Configuration Register
	
Normally PCR 0-7 are reserved for firmware usage. In coreboot we use just 4 PCR
	
banks in order to store the measurements. coreboot uses the SHA-1 or SHA-256
	
hash algorithm depending on the TPM specification for measurements. PCR-4 to
	
PCR-7 are left empty.
	
	
### PCR-0
	
_Hash:_ SHA1
	
_Description:_ Google VBoot GBB flags.
	
	
### PCR-1
	
_Hash:_ SHA1/SHA256
	
_Description:_ Google VBoot GBB HWID.
	
	
### PCR-2
	
_Hash:_ SHA1/SHA256
	
_Description:_ Core Root of Trust for Measurement which includes all stages,
	
data and blobs.
	
	
### PCR-3
	
_Hash:_ SHA1/SHA256
	
_Description:_ Runtime data like hwinfo.hex or MRC cache.
	

```

zaolin, 2019-03-27:
Cost estimation in man-days: 8 PD

closed 2022-02-21 by tlaurion