linuxboot/heads #502: Explore QubesOS AEM usb boot

tlaurion, 2018-12-28:


tlaurion, 2018-12-28:
Would be nice if [AntiEvilMaid](https://github.com/QubesOS/qubes-antievilmaid/blob/master/anti-evil-maid/README) could be supported from Heads, so S3 suspend and integrity validation would be implemented in QubesOS.

tlaurion, 2018-12-28:
It would complement Heads with memory measurements within QubesOS. Not sure if it would work with coreboot though, since latest AEM requires TXT. Will give it a try. Goal of it would be to boot from USB AEM disk from Heads.

tlaurion, 2018-12-28:
from @zaolin :

> Blobs of the original fw needs to be extracted. So searching with Uefitool might work for retrieving the ACM. I can help with that

tlaurion, 2019-01-20:
see [This WiP branch](https://github.com/tlaurion/heads/tree/x230_blobs). Unfortunately, I do not know how to extract what would be required so that SINIT would be functional.

@zaolin : ping! :)

tlaurion, 2019-01-30:
@zaolin updated:

> with TXT enabled stripping ME won't work
> use UEFI tool and text search with unicode enabled for "ACM" then extract the body
> 
> You should be able to extract the BIOS ACM from thinkpad vendor firmware
> acminfo of tboot tools gives you the output if chipset_acm_type equals BIOS and if the acm is valid
> 
> SINIT ACM != BIOS ACM

zaolin, 2019-02-04:
Stripped ME seems to work with TXT. We double checked that.

tlaurion, 2019-02-08:
#307 would be linked indirectly to this.

tlaurion, 2019-02-15:
@zaolin how you made it?

tlaurion, 2019-03-23:
Update from @zaolin: Waiting from Intel approval.

zaolin, 2019-03-27:
Cost estimation in man-days: 15 PD

tlaurion, 2022-07-01:
Well, #1172 provides TXT required ACM and SINIT blobs, years later. Where to go next?