linuxboot/heads #509: Multiple PGP cards

itay-grudev, 2019-01-18:
If multiple PGP cards are present, the `kexec-sign-config` script fails because it doesn't know which key/card to use.

Setting the `default-key` in `~/.gnupg/gpg.conf` should solve the issue. I don't know what's the best approach to solve this, so I won't do a PR.

But I suppose the `flash-gui.sh` script could be changed to set the default key id.

flammit, 2019-01-18:
@itay-grudev couldn't you set your custom `gpg.conf` by adding a CBFS file to the rom:

```
cbfs -o /media/x230.rom -a "heads/initrd/.gnupg/gpg.conf" -f /path/to/gpg.conf
```

tlaurion, 2019-02-16:
@itay-grudev : what is you use case of having multiple PGP cards in heads?

tlaurion, 2019-02-16:
@itay-grudev else @flammit is right, the best way would be to modify gpg.conf as desired, and then adding it into the rom with `cbfs -o /media/x230.rom -a "heads/initrd/.gnupg/gpg.conf" -f /.gnupg/gpg.conf` which would simply copy it over when [cbfs-init is called](https://github.com/osresearch/heads/blob/master/initrd/bin/cbfs-init#L19)

There is no danger in doing that.
You firmware would be modified. So in next reboot, a new HTOP/TOTP would be generated. And if you defined a default boot option to unseal LUKS slot 1 unlock key, you would have to set a new default to boot. But the changes of breaking your firmware from within through this are equal to none.

itay-grudev, 2019-02-17:
@tlaurion My use case is the following:
I have a YubiKey Nano constantly plugged in my computer. When turning on it causes problems when there is the YubiKey and the Librem Key.

It can be mitigated by just setting the default key in the GPG key import script.

closed 2024-04-16 by tlaurion