linuxboot/heads #516: How do you test heads with QEMU?

itay-grudev, 2019-02-08:
Kyle Ranking told me that when playing with heads one option to test is to simulate it with QEMU and some of you are using it. That will be much more convenient than re-flashing every time, but when I tried setting it up I couldn't.

Do you have a script or can you provide the qemu command with all the args and options you use?

pahakalle, 2019-02-08:
Is this what your looking for https://github.com/osresearch/heads-wiki/blob/master/Emulating-Heads.md

tlaurion, 2019-02-08:
@itay-grudev : I think make run does it, but wasn't able to simply [test it](https://gitlab.com/tlaurion/heads-ci/-/jobs/158449640).

[Emulating Heads](https://github.com/osresearch/heads-wiki/blob/master/Emulating-Heads.md) needs to be updated. Heads doesn't include Xen anymore [since kexec got fixed](https://github.com/osresearch/heads/commit/f618f09a6985a961a789ef52bc779d474ca3b947).

itay-grudev, 2019-02-08:
I'm compiling the `qemu-coreboot` image right now. Will give feedback later today.

tlaurion, 2019-02-08:
if anyone knows how to play around qemu in docker, that would be aweome to share.
[I innocently tried here](https://gitlab.com/tlaurion/heads-ci/-/jobs/158449640), without success.

tlaurion, 2019-02-08:
Another Heads [issue directs to bypass coreboot booting and instruct to boot kernel and initrd](https://github.com/osresearch/heads/issues/354#issuecomment-375697234)

tlaurion, 2019-02-15:
@itay-grudev how were your tests? :)

itay-grudev, 2019-02-15:
`make run` fails with:
```
❯ make run
qemu-system-x86_64 \
	--machine q35 \
	--serial /dev/tty \
	--bios /home/ito/Projects/heads/build/qemu-coreboot/coreboot.rom \
; stty sane
qemu-system-x86_64: Initialization of device e1000e failed: failed to find romfile "efi-e1000e.rom"
```

paulmenzel, 2019-02-15:
On 02/15/19 11:52, Itay Grudev wrote:
> `make run` fails with:
> ```
> ❯ make run
> qemu-system-x86_64 \
> 	--machine q35 \
> 	--serial /dev/tty \
> 	--bios /home/ito/Projects/heads/build/qemu-coreboot/coreboot.rom \
> ; stty sane
> qemu-system-x86_64: Initialization of device e1000e failed: failed to find romfile "efi-e1000e.rom"
> ```

As a workaround, please add `-nic none` to that call, and
check, if it fixes the issue. Otherwise, you need to find
the option ROM somewhere.

itay-grudev, 2019-02-15:
Thanks. I was able to get it working with:
```diff
diff --git a/boards/qemu-coreboot/qemu-coreboot.config b/boards/qemu-coreboot/qemu-coreboot.config
index 9427a2a..a0520fe 100644
--- a/boards/qemu-coreboot/qemu-coreboot.config
+++ b/boards/qemu-coreboot/qemu-coreboot.config
@@ -42,4 +42,5 @@ run:
                --machine q35 \
                --serial /dev/tty \
                --bios $(build)/$(BOARD)/coreboot.rom \
+               -nic none \
        ; stty sane
```

itay-grudev, 2019-02-15:
Do you know how can I get the GUI working? I am currently getting:

```
Error: Can't find usable screen
```

It's probably some QEMU option, I just can't find it.

tlaurion, 2019-02-16:
FYI: debian-9 can install qemu tools in a TemplateVM, fedora-29 can't for the moment. [It will be fixed in QubesOS 4.1](https://github.com/QubesOS/qubes-issues/issues/4805).

I gave up a long time ago trying to build Heads in debian templates, and[ CI based on ubuntu are currently failing](https://circleci.com/gh/osresearch/heads) @osresearch.

tlaurion, 2019-02-16:
> Do you know how can I get the GUI working? I am currently getting:
> 
> ```
> Error: Can't find usable screen
> ```
> 
> It's probably some QEMU option, I just can't find it.

https://github.com/flammit/heads/tree/qemu-gui-init

itay-grudev, 2019-02-16:
Yep. That works. I realised that some of the settings were missing, but didn't replicate them exactly. And that PR did it. Thanks a lot. I will close this and vote on your #527 PR.

@tlaurion Should we also create a PR with `-nic none` (diff above)?

tlaurion, 2019-05-12:
@itay-grudev Are you able to access an attached disk?

tlaurion, 2019-05-12:
@flammit used:
`
qemu-system-x86_64 --enable-kvm --machine q35 -m 1G --serial stdio --vga cirrus --bios build/qemu-coreboot/coreboot.rom -hda ../test.hd1.img --drive id=stick,if=none,format=raw,file=/home/flam/Downloads/Fedora-Workstation-Live-x86_64-29-1.2.iso -usb -device usb-ehci,id=ehci -device usb-tablet,bus=usb-bus.0 -device usb-storage,bus=ehci.0,drive=stick`

tlaurion, 2019-05-15:
Trying to find my way with qemu without KVM since QubesOS doesn't support nested virtualization.

[Didn't found a way yet to pass devices yet, which is problematic](https://github.com/osresearch/heads/issues/516#issuecomment-491631268), but I can at least test script changes without having to flash real hardware internally for everything.

gui-init requires an existing /boot. My testing requires a functional /media where oem-provisioning is found:
```
dd if=/dev/zero of=initrd/boot.img bs=1M count=1
sudo mkfs.ext4 initrd/boot.img
sudo mount initrd/boot.img /mnt/
sudo touch /mnt/oem
sudo umount /mnt
dd if=/dev/zero of=initrd/media.img bs=1M count=1
sudo mkfs.ext4 initrd/media.img
sudo mount  initrd/media.img /mnt
sudo cp oem-reownership/oem-provisioning.example /mnt/oem-provisioning
```

```
diff --git a/boards/qemu-coreboot/qemu-coreboot.config b/boards/qemu-coreboot/qemu-coreboot.config
index d670dc2..142c7f4 100644
--- a/boards/qemu-coreboot/qemu-coreboot.config
+++ b/boards/qemu-coreboot/qemu-coreboot.config
@@ -24,12 +24,12 @@ CONFIG_DROPBEAR=y
 
 #Uncomment only one of the following block
 #Required for graphical gui-init (FBWhiptail)
-#CONFIG_CAIRO=y
-#CONFIG_FBWHIPTAIL=y
+CONFIG_CAIRO=y
+CONFIG_FBWHIPTAIL=y
 #
 #text-based init (generic-init and gui-init)
-CONFIG_NEWT=y
-CONFIG_SLANG=y
+#CONFIG_NEWT=y
+#CONFIG_SLANG=y
 
 endif
 
@@ -40,15 +40,15 @@ CONFIG_LINUX_E1000=y
 
 #Uncomment only one BOOTSCRIPT:
 #Whiptail-based init (text-based or FBWhiptail)
-#export CONFIG_BOOTSCRIPT=/bin/gui-init
+export CONFIG_BOOTSCRIPT=/bin/gui-init
 #
 #text-based original init:
-export CONFIG_BOOTSCRIPT=/bin/generic-init
+#export CONFIG_BOOTSCRIPT=/bin/generic-init
 
 export CONFIG_TPM=n
 
-export CONFIG_BOOT_DEV="/dev/sda1"
-export CONFIG_USB_BOOT_DEV="/dev/sdb1"
+export CONFIG_BOOT_DEV="/boot.img"
+export CONFIG_USB_BOOT_DEV="/media.img"
 
 #run: coreboot.intermediate
 run:

```

tlaurion, 2019-05-15:
@osresearch : 

It also seem that from [gentoo documentation](https://wiki.gentoo.org/wiki/QEMU/Options#Machine) that it would be possible to pass host random stream by adding:`
-object rng-random,id=rng0,filename=/dev/urandom -device virtio-rng-pci,rng=rng0`

tlaurion, 2019-10-06:
To be tested:
https://github.com/tpm2-software/tpm2-tcti-uefi/blob/master/lib/efi-test-setup.sh

tlaurion, 2020-04-13:
As pointed in #700 :
`-netdev user,id=u1 -device e1000,netdev=u1` maps the e1000 to host interface and works.

Should be fixed in qemu-coreboot
@flammit : have a problem with that fix?

tlaurion, 2020-04-23:
So qemu-coreboot [TPM support](https://github.com/stefanberger/libtpms/wiki), which would rely on [swtpm](https://github.com/stefanberger/swtpm/wiki) support which depends on [libtpms](https://github.com/stefanberger/libtpms/wiki) support seems possible.

@orangecms: any success that path?

tlaurion, 2020-05-05:
A lot of documentation lies in this ticket for PR propositions. Reopening.

pietrushnic, 2020-05-14:
> So qemu-coreboot [TPM support](https://github.com/stefanberger/libtpms/wiki), which would rely on [swtpm](https://github.com/stefanberger/swtpm/wiki) support which depends on [libtpms](https://github.com/stefanberger/libtpms/wiki) support seems possible as dependencies of the qemu-coreboot board, but would be linked to a totally different build path depending if on fedora build system or debian (debian not providing libtpms packages for some reason, requiring to build everything from source)
> 
> @orangecms: any success that path?

@tlaurion I'm not sure what `qemu-coreboot` is, but no matter what emulated TPM using swtpm requires initialization by firmware, which for example is implemented in [SeaBIOS](https://github.com/stefanberger/seabios-tpm/blob/master/src/hw/tpm_drivers.c). Or maybe Linux kernel TIS/CRB driver is able to bring up TPM from scratch without firmware support - this should be pretty easy to check. Am I missing something?

tlaurion, 2020-06-18:
@pietrushnic : qemu-coreboot in Heads term is just the [board config](https://github.com/osresearch/heads/blob/master/boards/qemu-coreboot/qemu-coreboot.config)

So one can build Heads for qemu calling
`make BOARD=qemu-coreboot` or `make BOARD=qemu-coreboot-fbwhiptail`
and then call
`make BOARD=qemu-coreboot run` or `make BOARD=qemu-coreboot-fbwhiptail run`
and be able to test Heads, with really limited functionnalities, as exposed through #688 and #701 #354 . Any advice welcome.

ReLIFE9527, 2020-06-23:
> @pietrushnic : qemu-coreboot in Heads term is just the [board config](https://github.com/osresearch/heads/blob/master/boards/qemu-coreboot/qemu-coreboot.config)
> 
> So one can build Heads for qemu calling
> `make BOARD=qemu-coreboot` or `make BOARD=qemu-coreboot-fbwhiptail`
> and then call
> `make BOARD=qemu-coreboot run` or `make BOARD=qemu-coreboot-fbwhiptail run`
> and be able to test Heads, with really limited functionnalities, as exposed through #688 and #701 #354 .
> 
> Any advice welcome, since building those qemu board configs without them being linked to a software TPM (swtpm), not having functional USB passthrough (to be able to sign /boot components and remotely attest through HOTP) nor assigned prepared block devices/images attached (installed OS or at least a /dev/sda1 to be recognized by Heads as boot partition which is a requirement, sign /boot config with passed USB device) is pretty useless but to show that heads produces a valid ROM only.
> 
> qemu-coreboot boards have TPM deactivated, and as reported by issue #688, since no valid block devices are passed to qemu, the board is a diskless machine without proper changes on board configs themselves doing some more magic for the end user calling `make BOARD=qemu-coreboot-fbwhiptail run` to have measured boot, install an OS and have verified /boot as a Heads user would expect to be able to test out of the box.
> 
> @orangecms helped [here](https://github.com/osresearch/heads/issues/516#issuecomment-612721230). Maybe he could share some more light on the subject?

So the coreboot.rom or linuxboot.rom built by heads when I calling 'make BOARD=qemu-linuxboot run' is not able to boot the real kernel and root filesystem, since there is no disk can be mounted. Is it possible to build the necessary components separately and then combine them into the linuxboot.rom, and subsequently run it on QEMU with the real kernel and root filesystem?  Because I found QEMU can directly run the real kernel and root filesystem without linuxboot.rom, but when I added it, it would be stuck in the recovery shell, just as same as ‘make BOARD=qemu-linuxboot run’. Or, do you have some other good way to solve this problem? Sincerely thanks a lot.

tlaurion, 2020-07-18:
@ReLIFE9527 : I was successful applying some hacks [here](https://github.com/osresearch/heads/issues/516#issuecomment-492695279) while some help would be required to pass disks (or img) from the host to qemu so that the original mappings would be valid without hacking Heads board configurations but by having a [make run statement ](https://github.com/osresearch/heads/pull/707/files#diff-2c1abc05ab74ad59746fdc5e157bce85R59-R61) that works out of the box successfully: mapping usb devices (HOTP tests) with a swtpm being setuped correctly in a precedent step, with its required dependencies.

I am on unknown territories here. 
@pietrushnic any advices or name tagging here that would be helpful, or any link more [specific then the previously documented](https://github.com/osresearch/heads/issues/516#issuecomment-618194043) path that someone with more experience could shed some light on?

tlaurion, 2020-08-01:
Interesting USB usage tests were done [here](https://github.com/osresearch/heads-wiki/issues/32#issuecomment-553318317)

tlaurion, 2020-11-05:
@ReLIFE9527 @itay-grudev  @pietrushnic :  important breakthrough in code on the safeboot side for swtpm+qemu communication and board adjustments.

Reimplementing dependencies on qemu boards, Makefile and modules will lead to be able to test heads with TPM (and some additional tinkering would make USB pass-through of USB Security dongle work)

- [swtpm support in safeboot](https://github.com/osresearch/safeboot/issues/42)
- [relative commit](https://github.com/osresearch/safeboot/commit/6d60eef867b7a12ff185dd8ce87f68131d3858fb#diff-B67911656EF5D18C4AE36CB6741B7965)

pietrushnic, 2020-11-08:
@tlaurion thanks. We discussed that architecture and potential use cases extensively during [QubesOS minisummit SRTM talk](https://www.youtube.com/watch?v=Eip5Rts6S2I&feature=emb_title). Slides are [here](https://cloud.3mdeb.com/index.php/s/9XcFfrAmsEfd9xo)

tlaurion, 2020-11-09:
That was implemented as draft into https://github.com/osresearch/heads/pull/893 (doesn't build correctly as of now)

tlaurion, 2022-03-25:
For USB support (HOTP support) https://qemu.readthedocs.io/en/latest/system/devices/usb.html
Building instructions for swtpm missing in modules, so that qemu can ask for it and call it directly from make run.

PR #707 (2020-05-04): qemu-coreboot: finally a useable debug/test board
Now useful to debug something through make BOARD=qemu-coreboot.

TODO: [map a virtual TPM instance ](https://github.com/osresearch/heads/issues/516#issuecomment-538776741) and USB passthrough. 

Thanks to @orangecms for the tip. If you have others regarding https://github.com/osresearch/heads/issues/516#issuecomment-492695279, you are more then welcome.

Fixes #516 #701 and #158 

Based on top of #706 (should be merged first)

With testing happening on #700, another pull request will bonify he debugging output of this board too since debugfs wi [... cut]
closed 2022-08-25 by tlaurion