linuxboot/heads #517: Reduce  libslang.so.2 library footprint

tlaurion, 2019-02-08:
@kylerankin : doing some tests to reduce libraries footprints, since x200 and other hardware that don't have a TPM also tend to have smaller SPI flash chips (4mb on some, 8mb on others).

Whiptail depends on a 1.1M libslang.so.2 library, which is the biggest of all, with currently deactivated features:
```
--with-z=no \
--with-png=no \
--with-pcre=no \
--with-onig=no \
```

Can we deactivate some other unneeded features? Here is the list:
```
--with-terminfo=no \
--with-readline=no \
--with-slsyswrap=no \
--with-x=no \
--with-pcre=no \
--with-pcrelib=no \
--with-pcreinc=no \
--with-onig=no \
--with-oniglib=no \
--with-oniginc=no \
--with-png=no \
--with-pnglib=no \
--with-pnginc=no \
--with-z=no \
--with-zlib=no \
--with-zinc=no \
--with-iconv=no \
--with-iconvlib=no \
--with-iconvinc=no \
```

Edit: even if all of those are turned off, the library still pounds 1.1M. Any advice?

tlaurion, 2019-05-01:
@kylerankin : any idea?

tlaurion, 2020-05-03:
@MrChromebox : any idea here in link with #590 ?

MrChromebox, 2020-05-05:
@tlaurion unless I'm missing something, the only in-tree board I see including the slang module is qemu-coreboot (both explicitly and via CONFIG_NEWT).

tlaurion, 2020-05-08:
@MrChromebox and when whiptail is requested without full graphical interface (#453 ) to be able to use gui-init from command line without fbwhiptail for lesser space environements. (Not really a prority as of now).

those are combined space used per gui-init with fbwhiptail, current ticket being without fbwhiptail.
```
790103 832 -rwx------ 1 user user 851968 Jun 28 13:01 ./lib/libcairo.so.2
790113 652 -rwx------ 1 user user 667048 Jun 28 13:01 ./lib/libpixman-1.so.0
790114 200 -rwx------ 1 user user 201096 Jun 28 13:01 ./lib/libpng16.so.16
```