linuxboot/heads #531: GuixSD installer fails to boot

jfrederickson, 2019-02-20:
I was trying to install a copy of GuixSD on my x230 running Heads (with a build from master as of a few days ago). When selecting the USB boot option from the boot menu, I get this error:

[image]

There does appear to be something odd with the ISO; `/dev/sdc` in the following screenshot is the flash drive with GuixSD on it.

[image]

That said, I'm able to boot this installer image from another system without Heads installed.

tlaurion, 2019-02-22:
@jfrederickson I tested iso boot from usb.
you could try to do, from recovery: 
```
mount-usb
mount /dev/sdb2 /media
kexec-select-boot -b /media -c "*.cfg" -u
```

If you build master with this [commit](https://github.com/osresearch/heads/commit/aef5753373bf38854f24f0c88c6e42c627987ab5) in, you can change default usb boot partition to point to /dev/sdb2 instead of doing the above, and try again from gui-init (Graphical interface), or again, exit to recovery shell, modify /etc/config so that `CONFIG_USB_BOOT_DEV="/dev/sdb1"` to `CONFIG_USB_BOOT_DEV="/dev/sdb2"` and call `gui-init` to try again with that temporary configuration from the GUI

tlaurion, 2019-05-01:
@jfrederickson : status update?

jfrederickson, 2019-05-01:
I was able to generate an ISO from a running Guix system that booted successfully through Heads. I suspect there may have been a bug in the disk image tooling when the latest official ISO was generated that has since been fixed... in any case, I think the issue was pretty conclusively with the downloaded ISO and not with Heads.

closed 2019-05-01 by jfrederickson

daym, 2019-10-02:
Yeah, we (Guix people) have added some serious workarounds to the Guix system image after the 1.0.1 release.

As far as possible the previous version (including Guix 1.0.1) was as close to vanilla `grub-mkrescue` and `xorriso` as possible, which was a bad idea.

Thomas Schmitt (from `grub-mkrescue` upstream) helped us to use a sophisticated frontend that will work on more systems, especially buggy ones.

Long story short, (1) not sure it was a real problem on our side and we didn't just work around problems on your side and (2) should other distributions have a problem, they could try using `grub-mkrescue-sed.sh` from upstream as `--xorriso` argument to `grub-mkrescue`. We did and we are happy with it.

>I was able to generate an ISO from a running Guix system that booted successfully through Heads.

Since Guix is rolling release, then you used the new version mentioned above.